Jyers/Marlin #1400: 1.3.5 Firmware printing large model has layer shift

easonjim, 2021-09-18:
Dear,
   1.3.5 Firmware printing large model has layer shift，like this:
[image]
[image]
This is gcode file:
[aaalid.gcode.zip](https://github.com/Jyers/Marlin/files/7189440/aaalid.gcode.zip)
I have tried 1.2.2 firmware, same problem.
Printing small items does not present this problem.

Bukken5150, 2021-09-18:
Unfortunately this is not a firmware problem, but a hardware problem. Adjust the Belt Tension & Check the Pulleys as they are either to loose or to tight. More info about layer shift problem can be found here : https://all3dp.com/2/ender-3-pro-layer-shift-how-to-avoid-it/

tititopher68-dev, 2021-09-18:
@easonjim 
Like said @Bukken5150 , it's not a firmware issue but a hardware issue...

msdossajo, 2021-09-18:
@easonjim also check if the motherboard's fan is working during print, the way it is wired it will only work when the layer fan is working.

gothcha, 2021-09-19:
Do you hear a loud clunking during the print which coincides with the layer shifts? Some of the motherboards have a dodgy set of drivers that overheat on long prints. There are various discussions on the internet about this and improving the cooling to mitigate. From what I have seen there is no simple fix if this is your problem apart from a board change!

easonjim, 2021-09-22:
I collected some videos organized by netizens, and my phenomenon is very similar, I record here:
https://www.reddit.com/r/ender3v2/comments/kn8u72/layer_shift_investigation_data_and_video/

easonjim, 2021-09-22:
Through a lot of tests, I found that:
1. If the step number of E axis is set to 93, there will be no split-layer print
2. If the step number of the E axis is set to 127.4, staggered lines will be printed
3. If the step number of axis E is set to 140, there will be no split-layer print

Because the extruder of my ender 3 V2 has been modified to the double gear extruder of CR10 Pro, I have modified the step number of E axis.
I think it will be I modify the step number, resulting in the motherboard drive overheating, thus indirectly affecting the split-layer grain.