Jyers/Marlin #1402: Ender3 v2 V4.2.2 UBL 10x10 BL touch reboot when moving extruder

thecraftingdutchman, 2021-09-20:
### Description

When moving the extuder via the move menu, when reaching the end of the move the extruder moves in negative direction. and after a few moves the extruder stops moving. the screen freezes and the load bar forms an overlay in the screen and the printer is back in its fresh booted state.

### Steps to Reproduce

see description.

**Expected behavior:**  the extruded should move the relative distance provided in the menu, also with live movements enabled.

**Actual behavior:**When moving the extuder via the move menu, when reaching the end of the move the extruder moves in negative direction. and after a few moves the extruder stops moving. the screen freezes and the load bar forms an overlay in the screen and the printer is back in its fresh booted state.

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

tome9111991, 2021-09-20:
have a similar problem with live movement with Z axis. 

I turn to Z +100 when it reaches move negative sometimes.

tititopher68-dev, 2021-09-20:
@tome9111991 
In your case, this is due to the fact that the auto-correction of the mesh is active. Also in my build in development, I had made sure to deactivate the Mesh during Live movement ....

@thecraftingdutchman 
in your case, I don't have this issue (using the base of the last build and commit...)

tome9111991, 2021-09-22:
@tititopher68-dev 
I'm not sure it is from auto-correction of mesh.
Without live move i dont have the issue

https://youtube.com/shorts/pmPTHIYNGAk?feature=share

tititopher68-dev, 2021-09-22:
@tome9111991 
It's exactly what I said : no issue without Live move and issue with Live move. But with Live move and my correction (set level mesh desactivated), no issue