Jyers/Marlin #1408: SD card crash

aLAN-LDZ, 2021-09-21:
### Discussed in https://github.com/Jyers/Marlin/discussions/1407

<div type='discussions-op-text'>

<sup>Originally posted by **aLAN-LDZ** September 21, 2021</sup>
Hi

I have the latest version of your software installed. Everything was fine until today, I switched from the default to bltouch.

On the default software, I do not have this problem, on the bltouch software, when I click the first icon "print", the printer freezes and restarts.

when without an SD card, I first click on this icon and see the words no media and insert the card, I can see the files normally. Sometimes printing can be started.

I do not have this problem on the default software :( pls help</div>

tititopher68-dev, 2021-09-22:
@aLAN-LDZ 
Absolutely not a firmware issue, but either a motherboard issue (slot microSD...) or more simply, your microSD is the issue...So try to change your microSD...

aLAN-LDZ, 2021-09-22:
I can try a different card is not a problem, but as I say, if I use the firmware without bltouch everything is ok. Yesterday I reset the printer settings to the factory, after uploading the 5x5 hs bltouch and for now it's ok, when I get home I will try to turn on some printout and see

Cal-LF, 2021-09-22:
@aLAN-LDZ When you update firmware, ALWAYS clear eprom memory. Recurring issues will result if eprom is not cleared.

aLAN-LDZ, 2021-09-22:
Ok, if I can ask then what commands should I use for a complete factory reset?

I use m502

Cal-LF, 2021-09-22:
I use the reset offered in the menu of the firmware.

tititopher68-dev, 2021-09-22:
@aLAN-LDZ 
Not doing a Reset to Default (menu Control) after every flash -> possibility to have some issues....

Flashing a firmware with a bad microSD (FAT32 4096 bytes) -> possibility to have some issues...

Printing with a bad microSD -> possibility to have some issues....

"should the z coordinate on the screen change when printing the layer?" Yes

aLAN-LDZ, 2021-09-22:
after install stoc fw i dont have this problem, so its not a problem with sdcard :(

Cal-LF, 2021-09-22:
If this were a firmware problem, other users would be reporting as such. Reset eprom after every install.

aLAN-LDZ, 2021-09-22:
I already did, and the problem returned after a few power-ups of the printer. Tomorrow I will load the software again (I uploaded ori soft to the screen, maybe there was a problem

tititopher68-dev, 2021-09-22:
@aLAN-LDZ 
By Reset eeprom, we will say : 1/Reset to Default following by 2/ Store settings and 3/ Load settings...to initialize correctly the eeprom...

aLAN-LDZ, 2021-09-24:
I uploaded everything again, reset the eeprom as you say and unfortunately the problem continues - there is no problem on the original software. I have noticed that I am getting an error in octoprint console: SD Init Fail. Even if there is no physical card inside

tititopher68-dev, 2021-09-24:
@aLAN-LDZ 
Your SD Init fail is normal when no microSD card is inserted.
Where it is not is when it is inserted.  Either you have a problem with the microSD (you have to change it and format it in FAT32 4096 bytes and choose a microSD ≤ 16 GB, beyond that there may be problems and you need repair and in 16 GB), or your Motherboard has  a problem.
But it's not the firmware.  Once again the management of the SD and in particular the detection of the SD is made by the Marlin base common to all motherboards.
On the other hand, the management of the SD has evolved and changed since version 2.0.1 on which that of Creality is based.  Currently we are at version 2.0.9.2.
You must still doubt that if there was a firmware problem with the SD, 100% of users would have this problem, and fortunately, this is not the case at all ...
And I hope that you have well masked the 5v pin of your USB cable on the Rapsberry (Octoprint) side ... a source of various and varied problems.