Jyers/Marlin #1411: Z axis compensation does not work

aLAN-LDZ, 2021-09-22:
1) Install the latest software 5x5 or 5x5 hs
2) zero axis alignment - good
3) offset settings z - good
4) write to eeprom
5) I did auto leveling (g29)
6) I wrote to eeprom
7) I selected the auto-leveling options
8) at the beginning of the gcode I added: M501
M420 S1
9) Logs indicate that auto-leveling is on
10) when printing the first layer, the z axis will not even move by 0.01 (neither on the screen nor physically the stepper motor), as if the compensation would not work at all

tititopher68-dev, 2021-09-22:
@aLAN-LDZ 
Just for information, are you a beginner ?

Your startGcode is wrong.
At your point 9, what you are doing is wrong.
M501 is...the Reset setting Gcode...why ?
With this Gcode, you reset all the eeprom data's and...you delete automatically your mesh, your Z-Offset, etc...
Ok...
In your startGcode, no M501 !
You must have a G28 ; Autohome.
And after, if you did an autoleveling and stored the mesh, you add a M420 S1 ; 
If you didn't an autoleveling before begining your print, after the G28, add a G29 (following by a M503 if you want store the mesh in the eeprom).

It's not a firmware issue !
Sorry for my English, but I'm french....

aLAN-LDZ, 2021-09-22:
https://marlinfw.org/docs/gcode/M501.html acordting to this m501 load data from eeprom and does not delete them. Do I get something wrong?

tititopher68-dev, 2021-09-22:
@aLAN-LDZ 
I edited my last post (grrrr Google translate....)...
After every flash you have to do a Reset to Default (menu Control). Without that, your eeprom isn't correctly initialized. Your eeprom remains structured like the original one ... and it cannot work.  The settings will not be stored in the correct places ...

tititopher68-dev, 2021-09-22:
@aLAN-LDZ 
We are a lot of people using BLTOUCH 5x5 or 3x3 or UBL 10x10, etc... without issue with the Auto Bed Leveling...

Cal-LF, 2021-09-22:
@aLAN-LDZ These are not firmware issues. Your post would be better answered in the discussions area.

aLAN-LDZ, 2021-09-23:
> @aLAN-LDZ
> Just for information, are you a beginner ?
> 
> Your startGcode is wrong.
> At your point 9, what you are doing is wrong.
> M501 is...the Reset (load) setting Gcode...why ?

https://marlinfw.org/docs/configuration/configuration.html#bed-leveling

"For example, M420 S1 must be used after M501 to enable the loaded mesh or matrix, and to re-enable leveling after G28"

tititopher68-dev, 2021-09-23:
@aLAN-LDZ 
No... absolutely not necessarily to use M501 !!!
In the firmware, particularly in configuration.h, re-enabled after G28 is set by default...
And we repeat again, like @Cal-LF said, there are not a firmware issue ! And please, your post would be better answered in the discussions area.

closed 2021-09-24 by aLAN-LDZ