Jyers/Marlin #1417: Filament Change

Grenalis, 2021-09-24:
When doing a filament change, (in firmware), upon resuming, the X and or Y axis will slam in one direction, then the print will continue, approx. 5mm off where it should be on the Y axis.  This has happened a the last 2 times I did a filament change, of course requiring a restart of print without a filament change

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

tititopher68-dev, 2021-09-24:
@Grenalis 
Can you more accurate on the firmware version used, configuration files, etc.?
For several months, this Bug Marlin no longer exists, including in the Jyers's Firmware ...
This can not be a firmware issue, because the GCODE M600 is not modified by Jyers and is part of the common sources of Marlin ...

tititopher68-dev, 2021-09-24:
@Grenalis 
And see #1395

Grenalis, 2021-09-24:
The current E3V2-BLTouch-5x5-HS-v4.2.7-v1.3.5.b .  Happened both times I did a filament change via firmware.  Not using any specific config files.  Just the firmware.

tititopher68-dev, 2021-09-24:
@Grenalis 
But sorry, no firmware issue.  Filament change is managed in the firmware by the call to a Gcode M600 ... which has not been modified by Jyers and which has been coded by Marlin for all printers.
And as for @DJWonderfull, in the end no issue.  I just did it and no issue.
I take this opportunity to remind you that you have to compile your own firmware from the build and the latest commits here.  The Version 1.3.5.b is old (there have been no precompiled firmwares since ... Jyers being very busy with his move and reinstallation).
Another important thing, you must always after flashing the firmware, do a Reset to Default, following by a Store settings and following by a Load settings ...

tititopher68-dev, 2021-09-24:
@Grenalis 
Perhaps your issue is a Cura issue. Try if you can PrusaSlicer...

Grenalis, 2021-09-25:
@tititopher68-dev 
What is the newest Jyers if 135b is old?
I will try a filament change in cura.  I have always did it in the past via interface on printer... ie firmware