Jyers/Marlin #1420: Feature Request - Ender 3 V2

Cobra1502, 2021-09-24:
### Description

Make a filament usage display on the home screen that prints out how many meters of filament have been consumed since power up or reset of counter.

This may be a CPU intensive exercise? as it would require real time monitoring of the extruder stepper motor to see how many positive feeding steps less retraction steps have occurred.  This likely would require the unique setting each person has programmed into their extruder calibration.

Thank you in advance for considering this.  If it has been addressed else where let me know as my Google searches failed to show anything.

tititopher68-dev, 2021-09-24:
@Cobra1502 
Not possible, sorry...
And this has already been mentioned, there is no time in the Discussion section ...

tititopher68-dev, 2021-09-24:
@Cobra1502 
I give more details.  The length of filament used can be counted by Marlin but we cannot reset this counter (or any other in particular), just only all the stats.

tititopher68-dev, 2021-09-24:
@Cobra1502 
Displaying this value in the item info is quite easy.  Just like resetting the stats.

Cobra1502, 2021-09-25:
Are you saying it’s easy to do or already has been done. 

Tom Kindred

Cobra1502, 2021-09-25:
Thanks. 

Tom Kindred

tititopher68-dev, 2021-09-25:
@Cobra1502 
I already have been done in my own build in development.

tititopher68-dev, 2021-10-02:
@Cobra1502 
Small mistake on my part.  The length of filament used is already displayed in Control -> Info ...
Second line under Prints and Finished ...
Only the Reset Print Counter function is not implemented.

tititopher68-dev, 2021-10-02:
@Cobra1502 
[image]
[image]