Jyers/Marlin #1421: Feature Request: Filament Run Out Settings Display

Cobra1502, 2021-09-24:
### Description

My Google searches do show a few messages regarding the need to add lines and compile code in order to deal with certain filament runout sensors.  Why not add as a standard compiled feature the 2 or 3 display settings boxes required to set the Ender 3 V2 up to work with the most common run out sensors.  For example Beach Tree, Creality, etc.

tititopher68-dev, 2021-09-24:
@Cobra1502 
Because in 99% of cases it is not possible.  Marlin sets parameters which will only be taken into account once during compilation ...

The only possible thing that I could integrate into the interface, by tinkering with the management code of the STM32 CPUs, is the modification of the LOW or HIGHT mode of the detector.  For the rest, the only possibility is to modify the configuration.h file according to its detector and to compile its own firmware ...

Cal-LF, 2021-09-24:
@Cobra1502 Friendly note, feature request and ideas are better located in the discussions area. Issues area is for firmware problems.

Cobra1502, 2021-09-25:
Thanks for the heads up. New to this. 

Tom Kindred

eaytac, 2021-09-29:
@tititopher68-dev 
This feature requested before by @Legsmaniac in [this issue](https://github.com/Jyers/Marlin/issues/224). In this issue the request explained as;

> Recently I tried this firmware by mriscoc which has Filament Runout State in the Advanced Menu settings on the LCD. It has many settings very similar to Jyers and a few more.
> 
> It would be great if this feature could be added to Jyers because it simply cannot be beat.
> I noticed some other great features in mrisocs variant such as Selectable mesh points from 2x2 to 7x7, PID popup with temperature plot, Filament run-out sensor with active state setup (LOW/HIGH), Background Colour Change.
> 
> And before anyone asks why I don't go use mrisoc's firmware then - despite some cool features, I'm not convinced it's as good as Jyers, plus I get compile errors with his, I would much prefer to stick with Jyers, with or without these features. Just thought it would be nice if some, the filament runout state at least, could be included.
> 
> Thanks for reading and thanks for a truly great firmware.

Most of Jyers firmware users which has this problem will be very happy if "Filament Run Out" option added to menu like firmware by mrissoc. Waiting for your help, thanks for all your efforts and this great firmware.

tititopher68-dev, 2021-09-29:
@eaytac 
When I hear about this @mriscoc, it pisses me off.
This guy does not share anything, no real sources of these famous features...The ones he provides are bogus and do not correspond to what you have tested.  On the contrary, it does everything it can to erase all traces of code snippet concerning the functionalities.  On the other hand, to use part of the code of @Jyers or to correct the bugs of his firmware by referring to the work of @Jyers and others, without respecting the rules of the Marlin license, he is the best, a real champion of the  world !

Regarding this request, I am not @Jyers and he will implement it if he wants (but I very much doubt for having questioned him, a while ago, on the subject ...).  This goes against Marlin's policy...because it cannot be generalized to all motherboard printers.

For my part, I am finalizing my version based on the firmware of @Jyers but including some of the functionalities, including that of the filament detector, of this..@m ... c and others fontionnalities... and of course, all the sources  will be fully available.