Jyers/Marlin #1424: Bed leveling not working at all on 1.3.5b

JayBee-git, 2021-09-25:
### Description

Bed leveling isn't activating at all during prints on the last 1.3.5 firmware. It was working fine on older 1.2.2.

### Steps to Reproduce

1. Do a bed mesh and use I M420 S1 Z2 or use G29
2. Print

**Expected behavior:** [What you expect to happen]

The Z axis should change depending on the bed level

**Actual behavior:** [What actually happens]

The Z axis stays at Z offset for the whole print. No changes.

#### Additional Information

I did several resets/reinstallation. Nothing helps.

tititopher68-dev, 2021-09-25:
@JayBee-git 
I can assure you that this is not a firmware issue.  If it did, it would be a long time since 100% of users reported it.

The Auto Bed Leveling works perfectly fine, sorry.

Can you more accurate on the firmware version used, configuration files, etc...and describe your process to calibrate your printer ?

After every flash a new firmware, do a 'good' reset is : 1/Reset to Default following by 2/ Store settings and 3/ Load settings (menu Control)...to initialize correctly the eeprom...

And did you compile your own firmware with the correct config files and the latest build and latest commit here?

The latest pre-compiled version v1.3.5b is already old and does not benefit from the latest bugfixes from the latest sources here ...

tititopher68-dev, 2021-09-25:
@JayBee-git 
Just for you information...G29 is coded by Marlin for all 3D printers...not by Jyers.
M420 S1 S2 : Use the last mesh active and validated in the EEprom, auto-correction will be done only on the first 2 layers (parameter Z2) ...

JayBee-git, 2021-09-25:
> @JayBee-git
> I can assure you that this is not a firmware issue. If it did, it would be a long time since 100% of users reported it.
>
> The Auto Bed Leveling works perfectly fine, sorry.
>
> Can you more [...]

As I stated, I did a reset. And yes I know how to do a reset. I even reinstalled the older firmware, and everything works perfectly there.

I also tried M420 S1 S2 (it's actually the first thing I tried) but it didn't work either, so I first though my bed moved or something.
But no, the Z offset during print isn't changing at all like it should.

I will try compiling the latest source and report how it goes, thanks.

JayBee-git, 2021-09-25:
I compiled your latest source with your bltouch 3x3 template config file, did a full reset, full recalibration, full bltouch mesh (with preheated bed and hotend), tested print with either G29 and M420 S1, and I still get the exact same issue.

The Z axis is always staying the same along the whole first layer, causing small but noticeable thickness inconsistencies.
I suspect it's not caused by your firmware but the Marlin source itself, since all firmware I tested from 1.x source worked perfectly.

And no, I am not the only one reporting the issue, there are few other people around having similar issues (for instance on the smith3d site).

tititopher68-dev, 2021-09-25:
@JayBee-git 
Sorry to insist but if M420 S1 is not taken into account, it means that the mesh is invalid.  Either it has not been saved or it is not in the correct location of the EEprom and in this case it means that the EEprom is not properly initialized.  Also either the 3 operations necessary for the reset are not correctly carried out, or an EEprom problem (hardware), since the last version, it is almost used to the maximum, or a firmware which flashes badly and leads to malfunction.  ..
 The Z-Offset (fixed distance between the nozzle and the pin (trigged) of the BLTouch) does not move during printing.  It is the Z axis which makes positive or negative micro corrections ... but this cannot be seen on the screen, only by observing the Z axis coupler for example.

JayBee-git, 2021-09-25:
> @JayBee-git
> Sorry to insist but if M420 S1 is not taken into account, it means that the mesh is invalid. Either it has not been saved or it is not in the correct location of the EEprom and in this [...]

I understand, but with all the flashing and reset I have done, I am 100% sure I'm doing things correctly here.
I checked the mesh every time, and it seems good to me. It just seems to completely ignore it, since the difference of layer thickness are exactly the same as my bed slight tilt.
And again, the bed leveling does work perfectly with older firmware, both official creality firmware and your older 1.x versions.
And yes, that is the Z axis that stays at the exact layer height all along. You mean the onscreen Z axis value shouldn't change even with more than 0.1mm difference? It's usually showing the proper live value normally during the print.

tititopher68-dev, 2021-09-25:
@JayBee-git 
On the display, you see just the value of the Z-Offset. During the print, you see your Z value height corresponding to the different printed and current layers.  For example, in 0.2 ... at the first layer, Z -> 0.2, at the second Z -> 0.4, etc ...
But you will never see the positive or negative micro corrections displayed.
Also remember to recalibrate your extruder (100 mm technique).  And do an PID Auto-tune (there have been changes in the Marlin code on this point ...).  This has an impact on the grip and therefore on the first layer.
 If your Bed is almost flat, the correction will hardly be visible.

JayBee-git, 2021-09-25:
I already did all those calibrations as well. I am not an expert, but I am not a beginner either here ;)
Grip on the bed has no issue, and extrusion is perfect.

And my bed is almost flat as well, I did a manual bed leveling few times to ensure there were no big tilt of the bed.
But when printing level testing prints, I can easily see (and feel) the very small thickness differences appearing.
Just for testing, I did some print without the auto leveling, and I get the exact same results as with auto leveling.

And again, the auto leveling works perfectly with older firmware versions.

tititopher68-dev, 2021-09-25:
@JayBee-git 
In the firmware, the auto bed Leveling is carried out via the call to Gcode G29 (in bilinear mode) .. this Gcode is coded by Marlin, and not by Jyers.  In version 1.22 or 1.3.5b it is the same ...

JayBee-git, 2021-09-26:
> @JayBee-git
> In the firmware, the auto bed Leveling is carried out via the call to Gcode G29 (in bilinear mode) .. this Gcode is coded by Marlin, and not by Jyers. In version 1.22 or 1.3.5b it is the same ...

That might be, but the results are still here. 1.2.2 => leveling works, 1.3.5b leveling doesn't work properly.
I'm just telling you the facts, nothing less, nothing more. You do what you want with it.

tititopher68-dev, 2021-09-27:
@JayBee-git 
Sorry for your disappointments, but unfortunately I can only say that for thousands of people, it works ... The management of EEprom by Marlin is not the same on version 2.0.7 of 1.22 and version  2.0.9.2 current.
 In my opinion, your problem is hardware.  If the problem was the firmware, we would all, without exception, have your problem.  Which is absolutely not the case!  This is a proven fact.
 Unfortunately, we don't really know your configuration, what you have or haven't compiled ... and we don't have your printer in front of us.
 I'm afraid you don't have an answer.
 I'm just going to finish and not come back to the topic again, tell you that I had a user for whom some settings did not save in EEprom after an update ... I came with another motherboard  and there, everything was OK with the same firmware .... Then explain to me the why, if not a fault in the EEprom (write error verified later by noticing that the capacity of the available EEprom was continuously varying without doing anything ... I would have to find the piece of code that made it possible to highlight this ...).
 Now I can also make you do a shift and an increase of the currently restricted EEprom.
 In creality_dwin.h, change: size_t eeprom_data_size = 48 to: size_t eeprom_data_size = 64
 And recompile ...

MicroSur, 2021-09-29:
How we can be sure that M420 S1 taken into account during printing start?

cjmalchow, 2021-10-04:
> 
> 
> How we can be sure that M420 S1 taken into account during printing start?

In your slicer add that command to your start G-CODE and it should automatically include that and any other commands you've entered. There are many examples via a Google search. I'd also recommend ensuring you use slot 0 for auto bed levelling. This alleviates many problems ppl have with this firmware and ABL.

Failure2Society, 2021-10-07:
I am having the same issue with the Z axis not being homed when I put the printer into the homing section, I would post my current setup, but I have modified my ender 3 v2 to have the 400mmx400mm bed. my printer was working fine before I tried the v1.3.5 firmware compared to my old version, and the only issue is the z axis not wanting to home.

cmadon, 2021-11-03:
> @Failure2Society that issue is usually cause by choosing the wrong board version (4.2.2 vs 4.2.7) or not resetting eeprom after flashing.
> 
> @JayBee-git Did you end up resolving the issue? If not I'd recommend trying the latest source

I'm having similar issue as well.  I have never reset eeprom after update.  I have seen above the method...

"1/Reset to Default following by 2/ Store settings and 3/ Load settings (menu Control)...to initialize correctly the eeprom..."

However will this not cause settings to be lost and nothing to load back in the step 3 above?  I would think resetting to default then storing settings would cause default settings to be stored in step 2 then the default settings saved in step 2 would then be loaded back in step 3.

Am I missing something?

Jyers, 2021-11-05:
@cmadon You are correct, this will reset all your settings. Either save them with an M503 command to be re applied from the terminal, or write them down somewhere before resetting

cmadon, 2021-11-05:
> @cmadon You are correct, this will reset all your settings. Either save them with an M503 command to be re applied from the terminal, or write them down somewhere before resetting

Thanks! That's what I thought, just wanted to be sure I wasn't missing something.

Thanks again!

JayBee-git, 2021-11-13:
> @JayBee-git also interested. I seem to be having the same issue. Trying to track down the cause. Like you I have tested with and without G29 and there is no difference. Going to do some investigation of the debug output to see if I can find the cause.
> 
> I’m using the 5x5 abl options with ender 3v2. About the eeprom possibility. With just g29 it should be saved in ram and applied until power cycle or other code that resets correct? In my case I am not loading from eeprom with M420 after g29

Gave up on it, apparently issue is from the Marlin firmware that doesn't fully corrects the bed level from the bltouch mesh in some (rare) cases.
The mesh is there, it's active, loads correcly at each print, I can visualize it and it seems accurate and on point with the manual measurement I did by moving the hotend around and checking with a digital caliper.
It does try to correct the bed level, it just doesn't correct it fully, even tho the bed isn't tilted so much (less than +-1mm of height difference on corners).
I'm not saying it happens to everyone or is a general issue, it just did happen to me and was pretty much unfixable.
Compiled several different firmwares from various sources, even made my own custom firmware, cleaned the eeprom and reset the whole thing more time than I can count, and asked the Marling team about it as well.
The solution for me was just to redo a very long full manual bed leveling once again (without using the bltouch), and now it obviously works since my bed is perfectly leveled.
I have no idea where the problem came from, but at this point I don't care so much anymore, wasted too much time troubleshooting already.

tititopher68-dev, 2021-11-13:
@JayBee-git 
In 5x5 Bilinear mode, once the mesh is created, it is actually in RAM.  By validating "Save", it is saved in the EEprom.
The main problem in version 1.3.5.b is the handling of the SERIAL ports, the HAL of the STM32F1 CPUs of our motherboards and the SKR E3 mini V2 (which have the same problem).  This affects the operation of the BLTouch, the motor steps, the communication with the LCD, the card reader, the USB port ... Basically it messes up everywhere at random.
What needs to be done is to modify this new CPU environment back to the old environment that Marlin called Maple.
For that, I advise you to take this base there (v1.3.5.b + fixes):
https://github.com/Jyers/Marlin/archive/refs/heads/Deprecated-JyersUI.zip
You just modify the platformio.ini file.
You replace in line 16:
default_envs = STM32F103RET6_creality
by:
default_envs = STM32F103RET6_creality_maple
You can also increase the currently space of restricted EEprom.
In creality_dwin.h, change: size_t eeprom_data_size = 48 to: size_t eeprom_data_size = 64
And recompile your own firmware according to your choice of configuration's files.
You will have the same functions as v2.0.1 but without a lot of bugs, such STM32 issue (still existing in the base v1.3.5b)

tititopher68-dev, 2021-11-13:
@JayBee-git 
To complete and give an opinion, I do not believe that wanting to absolutely follow the Marlin Bugfix for our Ender3V2 is a good thing in the short term. The current Marlin releases have become unstable and are still "too young".
The most stable version (the one that I use and that I see maintained in all the fork of the firmware), is the Marlin base of the release v1.3.4.  According to me, @Jyers, with all the deepest respect I have for him, should have adapted v1.3.5b by making the necessary fixes, maybe also based on the organization of the files of the current release and define that as the stable version and consider the release  current as experimental or alpha.
I understand that it requires more time and investment but by opening a dev branch to volunteer developers, we could help him in this task ...

vkichline, 2021-11-16:
After weeks of experimenting with this issue on my printer, I found that:
M420 S1 ; Nothing happens, ABL does not adjust Z axis while printing
M420 S1 F10 ; ABL seems to be working perfectly
F is the fade level; marlinfw.org indicates that it's only used for UBL, I am using bilinear leveling.
Perhaps the default value for F is zero, and the ABL mesh fades to flat before the first layer is printed.
I am running E3V2-BLTouch-3x3-HS-v4.2.2-v2.0.1.bin.

MicroSur, 2021-11-16:
>  I advise you to take this base there (v1.3.5.b + fixes): https://github.com/Jyers/Marlin/archive/refs/heads/Deprecated-JyersUI.zip You just modify the platformio.ini file. You replace in line 16: default_envs = STM32F103RET6_creality by: default_envs = STM32F103RET6_creality_maple 

My fault or what?

`Compiling .pio\build\STM32F103RET6_creality_maple\src\src\lcd\e3v2\creality\creality_dwin.cpp.o
In file included from C:\users\user\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/WString.h:29:0,
                 from C:\users\user\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:47,
                 from C:\users\user\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30,
                 from c:\!3dprint\jyers\marlin-deprecated-jyersui\marlin\src\hal\shared\marduino.h:36,
                 from Marlin\src\lcd\e3v2\creality\../../../inc/../HAL/./STM32F1/HAL.h:32,
                 from Marlin\src\lcd\e3v2\creality\../../../inc/../HAL/HAL.h:30,
                 from Marlin\src\lcd\e3v2\creality\../../../inc/MarlinConfig.h:31,
                 from Marlin\src\lcd\e3v2\creality\rotary_encoder.h:32,
                 from Marlin\src\lcd\e3v2\creality\creality_dwin.h:30,
                 from Marlin\src\lcd\e3v2\creality\creality_dwin.cpp:32:
Marlin\src\lcd\e3v2\creality\creality_dwin.cpp: In member function 'void CrealityDWINClass::Menu_Item_Handler(uint8_t, uint8_t, bool)':
C:\users\user\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/avr/pgmspace.h:29:59: error: expected primary-expression before ')' token
 #define sprintf_P(s, f, ...) sprintf((s), (f), __VA_ARGS__)
                                                           ^
Marlin\src\lcd\e3v2\creality\creality_dwin.cpp:3660:19: note: in expansion of macro 'sprintf_P'
                   sprintf_P(cmd, PSTR("G29 J"),);
                   ^~~~~~~~~
*** [.pio\build\STM32F103RET6_creality_maple\src\src\lcd\e3v2\creality\creality_dwin.cpp.o] Error 1`

MicroSur, 2021-11-18:
Why are they recommending sources that cannot be compiled?

drphil3d, 2022-01-04:
I've been pulling my hair out trying to figure out what the hell is wrong. The Z does not move up until the layer height value is greater than the baby stepping value. I can adjust the Z up and down with babystepping but it never advances to the next layer. So I end up with 3 or 4 layers being printed in the same place with a z offset of -.55~
Bugfix JyersUI v2.0.1 Nov5 2021