Jyers/Marlin #1426: Problems Display and Print Abort

AR84115, 2021-09-25:
Hello.  I have an Ender 3 V2 with BlTouch and the current firmware on it.  When I cancel a print, the display freezes with the confirmation.  So printing is canceled but the display remains.  It also pulls back the entire filament.  and the nozzle moves instead of upwards and away in the direction of x0 downwards and in the direction of x0.  What can that be?

tititopher68-dev, 2021-09-25:
@AR84115 
Can you more accurate on the firmware version used, configuration files, etc.?
After every flash a new firmware : 1/Reset to Default following by 2/ Store settings and 3/ Load settings (menu Control)...to initialize correctly the eeprom...
And did you compile your own firmware with the correct config files and the latest build and latest commit here?
The latest pre-compiled version v1.3.5b is already old and does not benefit from the latest bugfixes from the latest sources here ...

tititopher68-dev, 2021-09-25:
@AR84115 
I can assure you that this is not a firmware issue.  If it did, it would be a long time since 100% of users reported it.