Jyers/Marlin #1455: Version 1.3.5 and 1.3.5b won't let me change UBL mesh slot and won't load slot 0 properly

edulpn, 2021-10-01:
### Description

First of all, thanks for the awesome and super stable fork! I recently acquired a Creality CR-Touch for my Ender 3 v2 4.2.2 board and started using the 15x15 UBL version of the firmware (pre-built `E3V2-UBL-BLTouch-15x15-v4.2.2` files). After some tinkering to get UBL to load the saved mesh I noticed the following:

- Versions 1.3.5 and 1.3.5b won't let me change the UBL slot;
- The above versions won't load the saved mesh if I either save it in the slot 0 and try to load the mesh from the default slot or use a `G29 L0` command;
- Downgrading to version 1.3.4 lets me change the mesh slot to either slot 0 or 1, but also doesn't load mesh properly from slot 0. I have to use the following start Gcode to load the mesh saved to slot 1:
```
G28 ; Home all axes
G29 L1 ; Load UBL Mesh at Slot 1
G29 A ; Activate UBL
G29 J2 ; Tilt the Mesh
```

Has this been noticed by someone else? Is there any way to determine if this is an issue with my specific machine if it was the case?

### Steps to Reproduce

It took me a while to understand the pattern and I almost didn't notice the issue with smaller prints in the center of the bed because I usually keep my bed pretty levelled manually.
To determine if a slot is selected I always select the slot, save/load the mesh and save settings to EEPROM from `Control` menu. After that I use the `Visualize Mesh` option under the `Level` menu.
I always generate meshes and save/load them from the `Level` menu, nothing done via Gcode scripts.

**Expected behavior:**
I expect to be able to use both mesh slots on all firmware versions, either from the `Level` menu or via `G29 L[SLOT]` Gcode commands.

tititopher68-dev, 2021-10-01:
@edulpn 
Not a firmware issue, sorry...why ?
Because :
1/ After every flash, do a Reset to Default following by a Store Settings following by a Restore Settings to initialize correctly the EEprom.
This is all the more important with version 1.3.5.b ... (see the next point).

2/ Since version 1.3.5.b, Almost the entire size of the EEprom is occupied.  Also the firmware calculates the space available for the slots in UBL.
And there, there is only one slot available on 0.

3/ The value -1 is normal.  Even if the slot is not fixed, it will still be saved internally in the 0 ... provided that the EEprom has been initialized before.

4/ We can improve the capacity of the EPProm, by modifying the creality_dwin.h file, the following value: size_t eeprom_data_size = 48 to size_t eeprom_data_size = 64.

5/ Of course, you have to compile your own firmware from the latest Jyers sources and taking point 4/ into account.

edulpn, 2021-10-01:
@tititopher68-dev Thanks for the insights, I was actually suspecting the extra slot was disabled/removed. But the docs/release notes don't mention anything in that sense, I thought I'd ask.

I'll do a full EEPROM init and try using slot 0 on my start Gcode. Will post the results and close the issue to add some history.
Since I don't actually need two slots right now, I'll just keep using the prebuilts, but I got the gist on expanding capacity.

edulpn, 2021-10-04:
So I followed the advice from @tititopher68-dev to do a proper EEPROM init and it solved my issues. Loading the mesh from slot 0/default slot now works with all firmware versions I tried. Closing the issue and leaving this as history for future searches!