Jyers/Marlin #1458: After 'Load filament' cursor on display only

ims21, 2021-10-03:
3v2 - 1.35b

Load filament
used "Cancel" for no more extrusion and finishing this option
use two returns (Back, Back) - it must be done in time, when Z still going down
on LCD will be after second displayed cursor only, without lines and then helps Off/On power only.

It can be simulated every time.

eacxx, 2021-11-04:
> Is it my problem only or no ?

No, it's not only you. We are two :)

Le0xFF, 2021-11-07:
And three, I have the same problem!

RoP09, 2021-11-15:
Have the same issue und 2.0.1 still

mapfeld, 2021-11-26:
Does it only occur occasionally, or all the time? I realized that after hitting 'cancel' in 'Load filament', I had to wait for the motion of the z-axis to stop **before** selecting any other item in the Menu. But this is rather a workaround than a solution.

ims21, 2021-11-26:
If you do not waiting, then always.