Jyers/Marlin #1459: X-axis crash

memomedfish, 2021-10-03:
### Description
First of all, thank you for your efforts in creating such a great firmware!
I am absolutely a beginner in 3D printing and at this point, I feel only comfortable using the precompiled firmwares.
I tried to install the 3x3 HS for 1.3.5b and I am getting this silly problem. Every time, I try to create a mesh, the extruder and fan assembly is crashing to the printer frame while its trying to reach the last probe position (this is also happening in 5x5 firmwares). I tried to play with probe and x/y offsets but no matter what I did, crash still happens.
I am using a fang fan duct with a chain bracket behind (that is the one crashing) which may be causing the issue.

I noticed that print area is marked as 230x230x250 in Jyers info, however it works totally fine with creality's 220x220x250.
Is there a way to fix that without getting into codes?

Thank you so much for your help!
Memo

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]
define print area as 220x220x250
**Actual behavior:** [What actually happens]
extruder head + fan assembly is crashing into printer frame
#### Additional Information
I am using a briss fang mod with a chain support.

tititopher68-dev, 2021-10-03:
@memomedfish 
Edit configuration.h and modify X_MAX value according to your briss fang mod...And compile again with the actual sources here.
The actual value is for the stock Hotend...

F-Light, 2021-10-07:
What development board should I choose? Mine is ender3v2. V4.2.2 is displayed on the motherboard.

> @memomedfish 编辑 configuration.h 并根据您的 briss fang mod 修改 X_MAX 值...并使用此处的实际源再次编译。 实际价值是为股票 Hotend...

memomedfish, 2021-10-07:
I used E3V2-BLTouch-3x3-HS-v4.2.2-v1.3.5.b After taking off the chain bracket, it all worked fine!