Jyers/Marlin #1469: Rendom reboots

FinlessBob, 2021-10-06:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

FinlessBob, 2021-10-06:
> @FinlessBob Sorry but no firmware issue...Hardware issue(s) ! After every flash, do a Reset to Default following by a Store Settings following by a Restore Settings to initialize correctly the EEprom. This is all the more important with version 1.3.5.b
> 
> Please post your request in Discussion area...

I did that day one as I do know this reset fact from long before. In fact after the first couple times I got a reboot, I did it all over again with reset, flash, reset, setup.
Could be a hardware issue but never happened prior to B firmware.  Thinking of rolling back one release as I did not have screen issues on the release before B anyway.

tititopher68-dev, 2021-10-06:
@FinlessBob 
if it was a firmware issue, just think that everyone, without exception, would have this outcome. Which is not the case. Sorry but absolutely a hardware issue...
(Try to compile your own firmware with the lastest source and commit here. The release v1.3.5b is too old). 

The versions 1.3.5 and 1.3.5b are in fact the same, except for the parameters modified in configuration_adv.h to limit certain thermal alarms -> On this subject, it is always necessary to make an Autotune PID to calibrate the temperatures and the temperature probes .

Badly flashed, corrupted firmware? .... Try to flash again (renaming it correctly) with another microSD (8 GB FAT32 - 4096 bytes)

tititopher68-dev, 2021-10-06:
@FinlessBob 
'What do you mean 1.3.5b is too old?' -> yes, because, since, there have been 2 Marlin bugfix merged, with a number of modifications of the management of the CPU and the serial interface among others.

Cal-LF, 2021-10-07:
A failing SD card or intermittent SD card socket can cause a reboot. An SD card extension cable can be troublesome.