Jyers/Marlin #1473: After manually stop prints

tzoltan56, 2021-10-09:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

gothcha, 2021-10-10:
I had the same issue, I modified my version of the code to add a disable steppers command, but not something simple to do unless you want to get deep into it!.

tititopher68-dev, 2021-10-11:
@gothcha 
Please can you contact me on Discord : Tititopher-dev#0331
Following for the firmware....

tititopher68-dev, 2021-10-12:
@gothcha 
Have you try this : 
void CrealityDWINClass::Confirm_Control() {
  ENCODER_DiffState encoder_diffState = Encoder_ReceiveAnalyze();
  if (encoder_diffState == ENCODER_DIFF_NO) return;
  if (encoder_diffState == ENCODER_DIFF_ENTER) {
    switch(popup) {
      case Complete:
        queue.inject_P(PSTR("M84"));
        Draw_Main_Menu();
        break;