Jyers/Marlin #1493: Documentation improvement

TomyCesaille, 2021-10-15:
### Description

> NOTE: I would have done it myself with a PR but it seems we [can't PR for the WIKI section](https://stackoverflow.com/a/11481887) in github.

I suggest to write in plain text the GCODE Scripts for OctoPrint along with the current screenshot. So, as users, we can copy paste it easily. Please notice my suggested content below.

Thank you for this outstanding work.

-------

(Content to add in <https://github.com/Jyers/Marlin/wiki/OctoPrint-Settings>)

### Before print job starts

```gcode
;Jyers gcode
M75 ;Start Print Job on Display
M117 <F>{{ event.name }} ;Send Filename to Display
```

### After print job completes

```gcode
;Jyers gcode
M77 ;Stop Print Job on Display
```

### After print job is cancelled

```gcode
;Jyers code
M77 ;Stop Print Job on Display
```

### After print job is paused

```gcode
;Jyers code
M76 ;Pause Print Job on Display
```

### Before print job is resumed

```gcode
;Jyers code
M75 ;Start Print Job on Display
```

Jyers, 2021-11-03:
Oh.. that's unfotunate... Well I suppose issues will have to do 😅 

I'll merge your changes in soon

Jyers, 2021-11-03:
Added! Thanks again :)

closed 2021-11-03 by Jyers