Jyers/Marlin #1558: "Display Off" Bug when printing.

CarboniteDan, 2021-10-30:
### Description

Display off feature does not work as expected.

### Steps to Reproduce

1. Start Print
2. Select "display off" menu option.

**Expected behavior:** [What you expect to happen]

- After "display off" option is selected, screen should shut off.
- Clicking the rotary encoder turns the screen back on.

**Actual behavior:** [What actually happens]

- Screen will not come back on after selecting "display off" option. Can hear beeps from menu selections. 
- Screen turns back on with encoder press once print is finished.

tititopher68-dev, 2021-10-30:
@CarboniteDan 
I confirm this issue.

tititopher68-dev, 2021-10-30:
@CarboniteDan 
Issue : yes but only if I print with Octoprint.  No issue if I print from SD from printer.
 The problem seems to be with the new Marlin encoder.cpp code in e3v2 / common.
 Marlin takes the wait_for_user variable into account.  This disrupts the flow of the code.  It must be the same exit for pause / resume #1551

tititopher68-dev, 2021-11-01:
@CarboniteDan 
For the moment, the time to fix all the issues,
I advise you not to take the base of v2.0.0, which is still far too unstable.
I advise you to take this base there (v1.3.5.b + fixes):
https://github.com/Jyers/Marlin/archive/refs/heads/Deprecated-JyersUI.zip
You just modify the platformio.ini file.
You replace in line 16:
default_envs = STM32F103RET6_creality
by:
default_envs = STM32F103RET6_creality_maple
And recompile your own firmware according to your choice of configuration's files.
You will have the same functions as v2.0.0 but without a lot of bugs, such STM32 issue (still existing in the base v1.3.5b), LCD glitch, LCD display OFF, correct position information, no more problems with long files, etc.

Jyers, 2021-11-03:
I believe I now have this fixed, please test to make sure it's working in your case too

Jyers, 2021-11-05:
This should be fixed in the latest release

closed 2021-11-05 by Jyers