Jyers/Marlin #1561: XYZ cords way off on the screen

Tapap84, 2021-10-30:
Z should be 1.46mm but it's reading 146.00mm. Same with other cords. Tried different files from different slicers and same results. 
[image]

tititopher68-dev, 2021-10-30:
@Tapap84 
Fixed in the last commit, 2 days ago, you have to recompile your own firmware ;-)
I advise you to learn how to do it ... Take a look here : 
https://github.com/Jyers/Marlin/discussions/1511#discussioncomment-1525774

Tapap84, 2021-10-30:
> @Tapap84 Fixed in the last commit, 2 days ago, you have to recompile your own firmware ;-) I advise you to learn how to do it ... Take a look here : [#1511 (reply in thread)](https://github.com/Jyers/Marlin/discussions/1511#discussioncomment-1525774)

You know what, I will. I've always been scared to do this but that doesn't look so bad. Thanks!

tititopher68-dev, 2021-10-30:
@Tapap84 
OK 👍 
However, I advise you not to take the base of v2.0.0, which is still far too unstable.
I advise you to take this base there (v1.3.5.b + fixes):
https://github.com/Jyers/Marlin/archive/refs/heads/Deprecated-JyersUI.zip
You just modify the platformio.ini file.
You replace in line 16:
default_envs = STM32F103RET6_creality
by:
default_envs = STM32F103RET6_creality_maple

You will have the same functions as v2.0.0 but without a lot of bugs, such as LCD display OFF, correct position information, no more problems with long files, etc.

Tapap84, 2021-10-30:
> @Tapap84 OK 👍 However, I advise you not to take the base of v2.0.0, which is still far too unstable. I advise you to take this base there (v1.3.5.b + fixes): [...]

Ok. i followed the directions and it created the .bin. Where did it save to?

tititopher68-dev, 2021-10-30:
@Tapap84 
in .pio\build\STM32F103RET6_creality_maple

Tapap84, 2021-10-30:
> @Tapap84 in .pio\build\STM32F103RET6_creality_maple

Cool. It worked. Thanks a ton. I'm watching videos now to learn more about editing. I was trying to do your flow change but that's still above what I can do

closed 2021-11-01 by Tapap84