Jyers/Marlin #1562: Exiting from the Tune Menu

rodolfopaulo, 2021-10-30:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

tititopher68-dev, 2021-11-01:
@rodolfopaulo 
For the moment, the time to fix all the issues, 
I advise you not to take the base of v2.0.0, which is still far too unstable.
I advise you to take this base there (v1.3.5.b + fixes):
https://github.com/Jyers/Marlin/archive/refs/heads/Deprecated-JyersUI.zip
You just modify the platformio.ini file.
You replace in line 16:
default_envs = STM32F103RET6_creality
by:
default_envs = STM32F103RET6_creality_maple
And recompile your own firmware according to your choice of configuration's files.
You will have the same functions as v2.0.0 but without a lot of bugs, such STM32 issue (still existing in the base v1.3.5b), LCD glitch, LCD display OFF, correct position information, no more problems with long files, etc.

Jyers, 2021-11-03:
Probably related to #1522 Give it a try with the latest source or if you don't compile try it with the next release

Jyers, 2021-11-05:
Release is out give it a try

closed 2021-11-05 by Jyers