Jyers/Marlin #1563: Print completes shortly after starting

Grenalis, 2021-10-31:
This is a bit different then the filename issue.  The issue i am seeing is that shortly after a print starts, it will go to complete on the screen, extruder stops, heaters turn off, and printhead continues, as if it was still printing.  If you dont power off the printer, it will continue to move all axis as if printing, but of course its not printing cause heaters and extruder are off.

Have you read Marlin's Code of Conduct? By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/.github/code_of_conduct.md

Do you want to ask a question? Are you looking for support? Please don't post here. Instead use one of the following options:

- The Marlin Firmware forum at https://reprap.org/forum/list.php?415
- The MarlinFirmware Facebook Group at https://www.facebook.com/groups/1049718498464482/
- The MarlinFirmware Discord Server at https://discord.gg/n5NJ59y.

Before filing an issue be sure to test the latest "bugfix" branch to see whether the issue is already addressed.

-->

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

tititopher68-dev, 2021-10-31:
@Grenalis 
Do you compile your own firmware with the latest Jyers's sources (commit https://github.com/Jyers/Marlin/commit/fe1eaec0c76e49bf729990eafa5c23f37ea62995)

However, I advise you not to take the base of v2.0.0, which is still far too unstable.
I advise you to take this base there (v1.3.5.b + fixes):
https://github.com/Jyers/Marlin/archive/refs/heads/Deprecated-JyersUI.zip
You just modify the platformio.ini file.
You replace in line 16:
default_envs = STM32F103RET6_creality
by:
default_envs = STM32F103RET6_creality_maple

You will have the same functions as v2.0.0 but without a lot of bugs, such as LCD display OFF, correct position information, no more problems with long files, etc.

Cal-LF, 2021-10-31:
@Grenalis  I have the compiled version that @tititopher68-dev recommended and it works with no issues so far.

Grenalis, 2021-11-01:
Sorry, E3V2-BLTouch-5x5-HS-v4.2.7-v2.0.0.bin

tititopher68-dev, 2021-11-01:
@Grenalis 
I advise you to learn how to do it ... Take a look here : #1511 (reply in thread)
One compiled, the file will be in .pio\build\STM32F103RET6_creality_maple
Don't be scared, it will turn out well and we will always be there just in case ;-)

Cal-LF, 2021-11-01:
@Grenalis My compiled version is for manual mesh V427 and will not work with your BLT.

Grenalis, 2021-11-02:
Much appreciated Cal.  I will give it a try

Jyers, 2021-11-03:
If you're no longer on the 2.0.0 source there's no reason to keep this open. If it comes back up in the future we can open a new issue and have a look into it

closed 2021-11-03 by Jyers