Jyers/Marlin #1574: Disable beep option missing?

bjoernbusch, 2021-11-04:
### Description

I pulled the latest code from the new branch and adapted the samples of the config files to my needs, which only involves a couple small changes. It compiled fine and the printer booted fine. The settings on the printer were lost / reset, so I tried adjusting everything again and I am missing the "Disable Beep" option. I checked the config files and the variable SOUND_MENU_ITEM was enabled. I tried commenting it out and compiled again, but no difference.

Should this be working already on the new branch?

### Steps to Reproduce

1. Pull bugfix-JyersUI 
2. Adjust config files and build and flash
3. Disable beep option should be available

If any additional info is needed, please let me know.

tititopher68-dev, 2021-11-04:
@bjoernbusch 
In configuration_adv.h, we have this (2 lines) :  

//Add a mute option to the LCD menu
  #define SOUND_MENU_ITEM

But this is misplaced because it is conditioned on : 
 #if HAS_LCD_MENU

Which is not the case for our DISPLAY.

These 2 lines must be deleted.

And add at the line 1283 : 
// Add a mute option to the LCD menu or DWIN E3V2
#if EITHER(HAS_LCD_MENU, HAS_DWIN_E3V2)
#define SOUND_MENU_ITEM
#endif

You have to compile your own firmware with this fix or wait for a new release....

bjoernbusch, 2021-11-04:
Awesome, thanks, the menu is back!

closed 2021-11-04 by bjoernbusch