Jyers/Marlin #1575: Ender 3 V2 hangs during change filament

R0B3RDV, 2021-11-05:
(ONLY WHILE CHOOSING LOAD FILAMENT - UNLOADING IS OK)

 I select load filament from menu, popup appears for temperture select and i press continue continue.... message appears that its waiting for the temperature;  as soon as it reaches temperature, extruder takes in filament.  after extruding the desired length (setting) , the display keeps hanging on 'waiting for processs to complete'... **I have to switch the printer completly off before it comes responsive again.** please advice

R0B3RDV, 2021-11-05:
i see that it shoud have been fixed in the current branch ?

tititopher68-dev, 2021-11-05:
@R0B3RDV 
Yes, this has since been fixed.  Either you compile your own firmware according to the choice of your configuration's files.  Either you wait for the next release ...
I invite you to learn (if it is not the case ...) to compile a firmware.  Between each release, there may be important fixes.

vhristenco, 2021-11-05:
@R0B3RDV so... the M600 works now?

tititopher68-dev, 2021-11-05:
@R0B3RDV 
YES !!!

Jyers, 2021-11-05:
Should be fixed in the latest release

closed 2021-11-05 by Jyers