Jyers/Marlin #1597: Visual problem icon display

mosheas, 2021-11-08:
Use : Ver 1.35 - Ver 2.0.1 - last compaile, 3x3 manual configuration. 
This happens after an command "change  filament ". from gcode printing, then I scroll the jog down to up 
And it is happened. 
Problem : An icon is not tied to its place on the top left corner. 
[image]

tititopher68-dev, 2021-11-08:
@mosheas 
Can you give more precision about : "This happens after an command "change filament ". from gcode printing, then I scroll the jog down to up"

mosheas, 2021-11-09:
When I finish replacing the filament start print , in Tune mode I rotate the button up and down to the end then the icon appears up on the left side