Jyers/Marlin #1600: M112 error on connect from Octoprint

plucker, 2021-11-09:
### Description
I can't connect to my Ender 3v2 with octoprint.  I reverted back to the 1.3.5 build and all is fine

### Steps to Reproduce
1. Power up printer
2. connect to printer with Octoprint connect button

**Expected behavior:** Oct print successfully connects to the printer

**Actual behavior:** Octoprint reports an immediate disconnect with attached messages in the log

#### Additional Information from Octoprint terminal
Trying port /dev/ttyUSB0, baudrate 115200
Connecting to port /dev/ttyUSB0, baudrate 115200
Handshake attempt #1 with timeout 2.0s
Connected to: Serial<id=0x6d1fb730, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=2.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: N0 M110 N0*125
Recv: \x0b\x9aÑ\x85ÑÕÍ\x81mismatch
Recv: Error:Serial status mismatch
Changing monitoring state from "Detecting serial connection" to "Error"
Send: M112
Send: N1 M112*32
Send: N2 M104 T0 S0*35
Send: N3 M140 S0*102
Changing monitoring state from "Error" to "Offline after error"
Connection closed, closing down monitor

tititopher68-dev, 2021-11-09:
@plucker 
Do you look at : https://github.com/Jyers/Marlin/wiki/OctoPrint-Settings

plucker, 2021-11-09:
I did look at that page and have all those settings correct and I have some insulating tape on the pin of the usb cable to prevent 5V coming in.

tititopher68-dev, 2021-11-09:
@plucker 
In v2.0.1 last commit ?

plucker, 2021-11-09:
That’s correct. I tried installing it twice, same result both times

tititopher68-dev, 2021-11-09:
@plucker 
Ok, I suspect a problem with MeatPack.
Do a test by commenting out this option in configuration_adv.h.
If this works, it means that the Marlin code concerning the management of MeatPack is in question.  Knowing that Marlin partially modified the code for the STM32 without maple ... this can generate this issue by reactivating the maple ...
This only confirms the loss of stability that existed with the Marlin base just before v1.3.5

plucker, 2021-11-10:
I tried that and the machine was OK with the Meatpack line commented out.  I'm just going to rebuild with it back in again to prove that that is the issue.

plucker, 2021-11-10:
Nope I uncommented it still works fine, so I put the downloaded release build back on and that works fine as well.  I don't understand, maybe I disturbed the tape on the 5V wire.  I'll close this anyway as all appears well.

tititopher68-dev, 2021-11-11:
@plucker 
You should have keep it open so that the side of @Jyers and especially Marlin, we look into the question.  Indeed, the instability of the Marlin code (and not the part provided by @Jyers) since their version 2.0.9.x, in particular on the management of SERIAL ports, essential in the operation of our motherboards with random issues.  As there are also with the UBL, the BLTouch, and sometimes reversals of motor steps ... The Marlin base of the 1.3.4 was and remains as it is the best for our Ender3V2.  It will one day become stable, but when?

plucker, 2021-11-11:
OK, re-opening it as it appears to be an intermittent issue.

ozonejunkieau, 2021-11-11:
For what it's worth, I am also having this issue with the exact same hardware. I did just identify then that powering up the printer with the OctoPi host disconnected and then plugging it in worked to "reconnect". I'm not sure how often this will work, but figured I'd let someone know that it may be related to the serial startup sequence?

bushybluejeans, 2021-11-11:
> @plucker 
> 
> Do you look at : https://github.com/Jyers/Marlin/wiki/OctoPrint-Settings
> 
> 
> 
> and did you mask the +5v pin of the cable ?

This fixed it for me (so far)

bushybluejeans, 2021-11-11:
Glad it's not just me. Getting the same M112 error with Octoprint on power up.

nike12000, 2021-11-15:
I have also been having this same issue and it also seems to be erratic and intermittent. It’s hard to prove but if I home all before I connect it “seems” like it does not happen as often.

BlueWings172, 2021-11-15:
I have the same problem ever since I upgraded to 2.01 firmware. I have [these Octoprint settings](https://github.com/Jyers/Marlin/wiki/OctoPrint-Settings) already.

Anyone figured a solution other than downgrading ?

bushybluejeans, 2021-11-15:
> I have the same problem ever since I upgraded to 2.01 firmware. I have [these Octoprint settings](https://github.com/Jyers/Marlin/wiki/OctoPrint-Settings) already.
> 
> 
> 
> Anyone figured a solution other than downgrading ?

I was having this issue recently. The steps I took to rectify it were:

1) I removed any unused and unwanted plugins from Octoprint.
2) I then Formatted the SD card and flashed the new firmware to the printer.
3) I then systematically did a Reset to Default, followed by a Store Settings and followed by a Restore Settings to correctly initialise the EEProm.
4) I also flashed the DWIN firmware to the display. Then followed the process in step 3.

pc-bastler, 2021-12-29:
I had the same issue with the 2.0.1 JyersUI... After 6-7 retries Octoprint connected to the printer.

Since I swapped out the display unit with a Creality touchscreen, I had to swap to the Creality-provided Firmware (also based on Marlin 2.0.1) and Octoprint is working as expected again.

tititopher68-dev, 2021-12-29:
@pc-bastler 
You should try with this version: https://github.com/Jyers/Marlin/discussions/1694#discussion-3749966
Either by compiling your own firmware with these sources and the associated configuration files, or by using the pre-compiled firmware provided.