Jyers/Marlin #1602: Autolevel Z-offset very inconsistent

plore-0, 2021-11-11:
### Description

Whenever I dial in the Z-offset and probe the bed, the Z-offset changes quite significantly (over 0.5mm). The temperature of the bed remains constant. I am using a CR-Touch and Jyer's latest firmware (E3V2-BLTouch-3x3-v4.2.2-v2.0.1.bin, occurs on 2.0 as well). Here are the steps that lead to my problem:

### Steps to Reproduce

1. Bring the nozzle down within a paper-width of the bed
2. Manually level the bed
3. power on
4. home the axes
5. go to Z-offset menu
6. enable live adjustment
7. microstep down until the nozzle touches the paper underneath (0.53mm)
8. save
9. go to Level menu
10. enable Leveling Active
11. Create New Mesh
12. return to Z-offset menu
13. enable live adjustment
14. microstep down until the the nozzle reaches the paper (1.03mm this time)
15. repeat steps 9-14
16. the Z-offset is now 0.93mm

**Expected behavior:** [What you expect to happen]
Z-offset should never change as the probe is fixed in place relative to the nozzle.

**Actual behavior:** [What actually happens]
Z-offset changes drastically.

#### Additional Information

Others seem to have this exact issue as well.

mrrstrat, 2021-11-12:
I have noticed that while the Ender 3 V2 is quite capable - remarkably - for making tight tolerance prints there are small repeatability issues with it inherent to the design and build. You are showing a variance in 0.1mm between step 14 and 16 and not sure what contribution the rotational sensors and minute slack in the mechanism plays in that as well. 

To shore up the mechanism I installed the dual Z axis kit that had a second servo and two bearing guides for the Z axis rods. I use digital calipers and rarely have noticed a difference in right and left ever beyond my own instrument and operator bias. So it appears to not cause more variation and prevents gantry sag.

I also have a CR Touch and have found results with setting the leveling with:

1. Making sure Z offset is 0.0, rehome axis.
2. In Z-offset menu enable live adjustment
3. Then after #2 use the option above it to land the print head on the board.
4. It won't touch the board after #3, but dial the height using the adjustment knob and micro adjustment to pass the paper test.
5. This is the new Z: save it.
6. Go to create new mesh and create one, and save it also to EPROM.

After #1-#6 are done, I assume I have what the printer believes the Z axis floor is and the Mesh that compensates for variation (I modified and compiled my own 5x5 firmware using the 3x3 code). I have been using the code option to use the stored mesh so my workflow is fast and I have had to remind Z and re-mesh after running the printer for 8-10 hours - if the first layer adhesion looks like it is not right.

So it looks like _the printer loses its fidelity from what is set and appears to be precipitous precision loss_ from the $100 is parts they use (but the results capable with this cheap hardware are nearly unbelievable!!).

Could you be seeing some precision loss and sensor resolution/accuracy loss (with a dash of amplified error)? If I understand your procedure it is a double calibration with the second one essentially performed to the last calibration baseline with the same system.

tititopher68-dev, 2021-11-12:
@plore-0 
Just for your information, there is a problem with the CRTouch.  The Z-Offset is the gap between the nozzle and the trigged pin.  The Z-Offset for the CRTouch, to function in the best way, must be between 1.5mm and 2.6mm also it is advisable to lower the position of the CRTouch by inserting washers or by printing spacers so as to be within the tolerances  requested.

mrrstrat, 2021-11-12:
I did not know that - where can I find details on this to correct it?

tititopher68-dev, 2021-11-12:
@mrrstrat 
Already, it is necessary that the geometry of the printer and the settings of the eccentrics of the vSlots wheels of the Z axis are correct.

There are no official docs.  Other users have already asked me the same problem and I, for my part, questioned the technical service of Creality.
The values ​​I give are theirs, which amounts to a gap between the nozzle and the Pin position at rest, 2.7mm to 3.8mm ...

By applying this, everything was back to normal.

Note that in your current configuration, I think that the accuracy of the CRTouch should not be good (approx. 0.03 ... instead of 0.003!).

My personal opinion is that a good old BLTouch v3.1 from Antlabs is much better than a CRTouch.

plore-0, 2021-11-12:
@tititopher68-dev 

To prevent confusion, I'm just going to refer to all Z-offset values as negative, which they are. Wouldn't lowering the CR-Touch as you suggest _increase_ the value of Z-offset? If my offset is currently ~-1mm, adding washers would make it be closer to 0mm, correct? Also, I'm not understanding how the relative position of the probe could affect its precision, as long as the nozzle isn't impacting the surface.

mrrstrat, 2021-11-12:
I guess I am assuming the position of the Z Axis floor is more or less _indeterminant_ to the printer once we remove the old Z Asis switch. In that case, I was thinking that the CR-Touch finds that safe distance between the position of its point and the actual nozzle. 
That's why setting the Z-offset made sense: it is the determination of the actual distance the nozzle needs to be from the plate (via the paper test). Then the re-forming of a mesh would be the scaler differences to apply at each zone on the bed to keep the nozzle at the perfect height for the first layer (at 0.2mm in my case with my layer height at 0.2mm).

In my limited experience with these printers - mere months - the auto leveling handles the issue of keeping the nozzle at a consistent height above the bed across the X,Y but the Z is indeterminant and can differ due to minute changes in where physical Z is. In my mind solving where Z is automatically would take a precision installation of the CR touch within a resolution of the thickness of a piece of paper. Seems so much easier to level the bed at one point for the center point to get Z.

If that's true, lowering the CR touch merely would lower where the indication of surface happens (effectively raising the bed to the probe/nozzle).

I might totally be missing something here but if this is how the CR touch and Jyers (marlin) firmware works then the OP might be confusing the first manual indication with a second run and be seeing some variation the second time around. That is a common problem in manually indicating certain types of points from a known start (and then chaining them along the way). You get also the benefit of error accumulation at each step along the way and the resolution of each effective point now is subject to more and more variation.

plore-0, 2021-11-12:
> I might totally be missing something here but if this is how the CR touch and Jyers (marlin) firmware works then the OP might be confusing the first manual indication with a second run and be seeing [...]

I thought that the mesh built from the autolevel probing only took effect in compensating the Z-height during prints and that the Z-position would not be compensated when just moving the head around in the prepare menu.

mrrstrat, 2021-11-12:
Plore-0 - That is a good question: I am assuming the position of the Z-axis is immutable once set and that the mesh provides a map to calculate differences from the known Z to determine where the bed is.

We *might* be saying the same thing as the Z becomes a floating point between the determined Z (via paper test) and the  range of values from the mesh.

Big thanks to titotipher: the eccentric nuts on my print head have much wobble. I am adjusting that. Gantry and all otherwise are good by chance as I adjusted nothing when I assembled the printer.

Is the OP having an issue seems to be based on understanding how Z axis determination is performed - I am not fully convinced I understand how it works _as the OP procedure was much different that I saw in those slick CR Touch installation videos by the Youtube stars_.

tititopher68-dev, 2021-11-12:
@plore-0 
I mean lower the CRTouch to have a Z-Offset value in the range of -1.5 to -2.6 mm.
With a Z-Offset of -1mm, we are just at the trigged / not trigged state change value for the CRTouch Pin.
As the CRTouch is taller than the BLTouch, according to Creality, the Pin in the upper position should be almost at the same level as the nozzle (while being a little lower than the Pin, about 0.5mm).
As the Pin is not adjustable like on a BLTouch, the slightest defect in the assembly and it becomes difficult...

tititopher68-dev, 2021-11-12:
During an Autohome, the mesh is automatically deactivated in the execution of the G28 and reactivated once done.

The Z-Offset does not intervene in the creation of Mesh.  It is only used to obtain the real position Z0 of the nozzle ...

The important thing is to do all this at printing temperatures.

plore-0, 2021-11-12:
I'd also like to know why none of the mesh values are 0 when a mesh is constructed and you look in the viewer. I thought the center square should be defined as 0 since that's where the z-axis is homed. What are the values relative to?

tititopher68-dev, 2021-11-12:
@plore-0 
Having a value exactly zero is hardly possible.
The mesh is not calculated from the point X and Y of the taking of the position Z0.  It is calculated according to a grid and margins.  This means that the Z0 position is almost all the time between 2 points of the mesh grid.  As the so-called bilinear mathematical compensation mode is used, a curve is defined to join these points.  Depending on the curve, we can therefore obtain positive or negative compensation values ​​... unless your Bed is perfectly flat and in a perfect geometric position ...

mrrstrat, 2021-11-12:
Case and point why I switched to a 5x5. The resulting compensation matrix really becomes an interpolation using the points of the bed and I figured a 5x5 was more accurately representing the surface with reasonable accuracy. 3x3 using a linearly connected grid would lose some of the function nature of the surface.

From what I see it looks like the mesh points Z values differ with the bed surface and do not use the Z offset in this determination (let me know if this is NOT the case).

I guess I am struggling with the lowering of the CR Touch to the bed - is this recommended or can the compensation work happily without this? My CR touch came with three different brackets and I used the one for my model printer.

> @plore-0
> Having a value exactly zero is hardly possible.
> The mesh is not calculated from the point X and Y of the taking of the position Z0. It is calculated according to a grid and margins. This [...]

mattpker, 2021-11-13:
@tititopher68-dev The "lower the CRTouch to have a Z-Offset value in the range of -1.5 to -2.6 mm" doesn't make any sense. All the CR Touch is doing is telling the firmware when it touches the bed, it should not care at all how far away the probe is from the bed.

I am using the CR touch with E3V2-UBL-BLTouch-10x10-HS-v4.2.2-v2.0.1.bin and having no issues using a z offset of -0.92.

I did this with the following steps:

1. Auto home
2. Enable live z offset
3. Use a piece of paper to figure out the z offset (for me it is - 0.92)
4. Go to the leveling, create new mesh
5. Save the mesh
6. Print something, make sure the following two g codes are after the G28: `G29 A` `G29 J`

plore-0, 2021-11-14:
Unfortunately, and quite expectedly, lowering the probe to be in the range of -1.5 to -2.6mm made no difference in the consistency of the z-offset. I'm still having to change it quite often.

tititopher68-dev, 2021-11-14:
@mattpker 
Contrary to what you say it makes sense.  Open a BLTouch (a CRTouch is built according to the same principle), and you will see a magnet at the end of the Pin and a detector on the body side (to put it simply).  There has to be straightforward detection for this to work.  With less than 1mm, this will not be the case.  Hence the values ​​communicated by Creality, unless he does not know their own product ...
In addition, your startGcode for the UBL is wrong.  After a G28, you need a G29 L0 then a G29 A and finally a G29 J2 ....but in ABL, after a G28, either a G29 or a M420 S1...
And before a G28, add a : 
M280 P0 S160 ; Reset all BLTouch alarm
G4 P100 ;

@plore-0
Another aspect to consider is the geometry of your printer.  Check the eccentrics of the vSlots wheels of the Z axis, that the height of the X axis is the same on the left and right side.  a difference greater than 0.4mm and that distorts everything.  Clearances at the screw to be checked.  The position of the helical screw must be perfectly straight and centered ... That your Z axis profiles are perpendicular, that the X axis, by making it go up and down manually, is not hard points, which  would mean that the vertical profiles are not parallel ... and yes, mounting your printer is a very important point.  There are videos on YouTube, in French, which explain and detail all this, Anglo-Saxon Youtubers do not like to show this too much (???).  Finally do your Z-Offset calibrations, mesh at printing temperatures ...
Good luck et cela ira (in french in the text) !

mrrstrat, 2021-11-15:
I guess I do not understand the major differences of the BL and CR touch. I know they work a little different. My CR touch is mounted to spec and with the correct bracket. Additionally it appears to indicate properly and detect values. This must at be the triggered values you are referring to.

Natelicious, 2021-11-18:
@plore-0 Im having a similar issue as you but I'm going to recheck all the screws. We may have not installed all the screws well.

During my test, I used the manual leveling feature and probed the bottom left spot. After each probe, the value kept increasing.... and i didnt see any compensation after using G29.

mrrstrat, 2021-11-18:
Now that this thread has been hijacked:

I am getting consistent and repeatable results once I flatten the bed and then set Z offset, then mesh:

1. I manually leveled the bed using the paper test at each corner in the Jyers firmware - forgetting about the middle point for now. 
2. Then using the SET Z OFFSET menu I reset the Z offset to zero and re-oriented the Z landing with the newly leveled bed (with live updates checked). 
3. Then I adjust the Z offset to the paper test (nice tug but not enough to bow the paper when pushed to the extruder).
4. I save the Z Offset 
4. Then re-do the mesh (saving the mesh). 

 I have been able to get nice flat beds with lots of black zones this way. And I am running print jobs with the stored mesh values.

I think if the Z offset "changes" its because of the loss of precision in the springs, wheels, and Z axis screws. If the bed is flat enough, only an occasional re-find of the Z offset and then remap the bed mesh makes the workflow pretty fast and reliable. I am using 25 points and use the stored mesh knowing it only needs to be re-mapped once in a while.

So back to this thread: is this issue an issue or a collective misunderstanding of the mechanism and the expected firmware behavior?

tekgamerpt, 2021-11-24:
Hello
I use a Capacitive sensor and i suffer from same problem...
I preheat hotend and bed, adjust the offset using paper but after a while.... For example, 4 or 5 home z axis, offset value previously choosen is not usable anymore. I check everything on printing and everything is fix, no loose parts...

I dunno why this happens but i can tell you for sure that no changes of bed were made between probing tests.

Hope this can help because i see that here everyone use CR or BL touch

ptheofan, 2022-03-02:
> 

Do you auto tilt after creating the mesh?
Do you create a plane out of the mesh after mesh and before tilting (or after tilting of you auto-tilt)?

And about the slicer G codes after G58, what codes do you use?
I have them like this
```
; UBL Initializer
G29 A ; Activate UBL.
G29 L0 ; Load 0 slot mesh from EEPROM.
G29 F10 ; Fade corrections for the first 10 mm (1cm)
G29 J2 ; Get the bed tilt in a 2 x 2 grid
; EOF UBL Initializer
```

mrrstrat, 2022-03-02:
I usually just create a mesh and then save it (using the 420 command in the slicer to use it and not re-probe before each print. I let the print first layer adhesion and subsequent layer adhesion tell me how much beyond that I need to go.

I have not explored UBL fully as I have not needed it. As you know to dial things in requires lots of time and filament :-).

I can tell you that the autotilt appears to be designed to compensate a mesh AFTER it is created. This compensation allows you to adjust the mesh stored rather than re-create with precipitous loss of bed level. It also gives you the ability to fine adjust after a mesh build.

Most of the parts I have made are within the zones of in-spec areas of the mesh's I have built and used: I do get bright green and red areas on the corners on occasion I ignore if they are significantly outside of the print area I am working in. But on my modified Ender 3 Max (pro V2) if I used the bed X,Y fully I would care more meticulously about the bed in its entirety. This is because not doing so on a large Z-axis print would certainly magnify these differences to cause layer adhesion issues later in the print.

Since I have not done any helmet-sized pints on my big scale printer, I have not "needed" to dial in using fully UBL - I have used enough to get results at the scale I have been using my gear at.

With that being said a great vid I use (with great recourses on the topic) for classes I teach on additive manufacturing provide a nice overview and application of the UBL. I have a 4th series I am planning that fully explores UBL for absolute resolution of bed leveling and layer adhesion issues.

https://www.youtube.com/watch?v=KA8URUR000Q

Great question and reminds me to get UBL implemented fully in my own work with my gear!