Jyers/Marlin #1603: Flow Rate and Leveling Active: not stored

Julesforman, 2021-11-12:
The flow rate value is not stored in the EEPROM after turning off the printer.
AND
The "Leveling Active" option does not remain checked after turning off the printer.
Both options, modified and configured, do not persist after turning off the printer despite saving the "Store Settings" option.

tititopher68-dev, 2021-11-12:
@Julesforman 
First of all the golden rule!  After flashing a firmware, you must systematically do a Reset to Default, following by a Store Settings and following by a Restore Settings to initialize correctly the EEProm.
(On this subject and for my own build, I implemented a functionality forcing the user to these actions, if the firmware has just been flashed).

Regarding the FlowRate, this behavior is normal and wanted by Marlin.  Each time the power is turned on, it will be reset to 100%.

Finally, concerning the "Active Leveling", it remains well memorized if of course the Mesh is valid and if we make a Store Settings (and of course, if we have applied the golden rule ...)

All this has already been mentioned many times and applies (for the moment) to all Releases.

Julesforman, 2021-11-12:
@tititopher68-dev
Thank you very much!

erstert3st, 2022-04-22:
"Regarding the FlowRate, this behavior is normal and wanted by Marlin. Each time the power is turned on, it will be reset to 100%."
Is there a way to bypass that ?