Jyers/Marlin #1609: Selector knob steps aren't registered properly

malinhiles, 2021-11-15:
### Description
Selector knob steps aren't registered properly.

### Steps to Reproduce

1. Use selector knob to move through menu items and change numerical parameters
2. Notice how the UI selector often does not respond to knob movements when backing out of menus and changing parameter values.

**Expected behavior:** 
UI responds to each notch from selector knob

**Actual behavior:** 
UI often does not respond to movements made on the selector knob. Response is inconsistent. Usually occurs when changing directions with the rotary encoder, UI selector wont move. This issue does not occur on stock Creality firmware.

Here is a video of the issue: https://drive.google.com/file/d/1tbMJqdk8ILsQrOMekqjz_sggFeXl_Bj6/view?usp=sharing

Additional notes:
 - Ender 3 v2 is less than a year old, and did not have the described issue on the stock firmware.
 - 4.2.2 Mainboard
 - Printer is stock (other than the Jyers firmware)

tititopher68-dev, 2021-11-15:
@malinhiles 
It has always existed for a long time with this firmware.  The firmware runs in a loop and the more menus and functions there are (much more than on the poor stock firmware), the longer the cycle times and the less the encoder follows ...  From a single step jump to the change of direction of rotation et not every time (this has already been mentioned in the past)!  If there is more, to have known it, it is rather an incipient failure of the encoder (which I replaced easily).

malinhiles, 2021-11-15:
Understood, so this issue doesn't happen on stock firmware because the stock firmware has considerably less features, and therefore has fast cycle times?

tititopher68-dev, 2021-11-15:
@tititopher68-dev 
This is indeed what we can consider especially since the code is the same in the operation of the encoder whatever the firmware ... code provided by Dwin and Creality

mdangschat, 2021-11-26:
@malinhiles do you use OctoPrint? I had a similar issue with the same printer hardware and Jyers 2.0.1. But it appears to have resolved itself after setting it up properly: https://github.com/Jyers/Marlin/wiki/OctoPrint-Settings

malinhiles, 2021-11-28:
@mdangschat No, I don't have any octoprint setup on this e3v2 machine. Its completely stock, good idea though! I've since moved back to stock firmware and my selector knob skipping anxiety has since reduced lol

sub256, 2022-01-05:
I came looking to see if anyone else experienced this. Machine is barely two days old so definitely not a failing rotary encoder. Hopefully this can be resolved somewhere along the line. I know it's a weird thing to be a deal breaker, but I couldn't deal, flashed back to stock.

mikecheng2626, 2022-03-21:
I have similar issue.
I flashed the latest Jyers display firmware, was a success. Flashed latest printer firmware E3V2-BLTouch-3x3-v4.2.2-v2.0.1, was a success. Rebooted. I was able to move around with the knob and did "all homing." Once the homing was done, I could still move the knob around, but I could not hear the clicking sound or click on any sections anymore. After that, I tried flashing display and printer firmware again, no success.
What should I do?