Jyers/Marlin #1612: M206 - Home Offset for Y not working

DerT94, 2021-11-16:
### Description

I wanted to set an offset for Y via the M206 command. Unfortunately, no matter what I enter here, it has no effect.

Commands used.
M206 Y-9
M500

M503
Recv: echo:; Home offset:
Recv: echo:  M206 X0.00 Y-9.00 Z0.00

### Steps to Reproduce

use the above commands.

**Expected behavior:** 
That the Y axis moves 9mm further away from the end stop

**Actual behavior:** 
The Y axis stops at the end stop

#### Additional Information

I use the pre compiled image "E3V2-Default-v4.2.7-v2.0.1.bin".

tititopher68-dev, 2021-11-16:
@DerT94 
Not a firmware issue...look at : https://www.youtube.com/watch?v=8MQRxBISYmU

DerT94, 2021-11-17:
> @DerT94
> Not a firmware issue...look at : https://www.youtube.com/watch?v=8MQRxBISYmU

Thank you

closed 2021-11-17 by DerT94