Jyers/Marlin #1624: E3V2-UBL-BLTouch-10x10-v4.2.2-v2.0.1

natewils, 2021-11-20:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Silenstryke, 2021-11-21:
I also experienced this with the same firmware using CRtouch, same point 42 of 100. 
Probe retracted after 41 and nozzle crashed into bed.

alexanderhenne, 2021-11-22:
I had a similar issue with the CRTouch where it would randomly fail a probe, while using that exact firmware. What fixed it for me was moving the CRTouch cables for OUT and G (the two top ones) to the Z stop header instead and compiling the firmware myself. No issues since. https://github.com/alexanderhenne/Marlin/commit/02d7b81baff282ed938fe2d154ba85f5a0ffe3e1

Naomarik, 2021-11-28:
I haven't had this issue using a newly purchased CRTouch on same firmware.

Sirnut, 2021-12-06:
I have this same issue with a CRTouch. I also believe it makes my meshes very inaccurate

YClayton, 2021-12-15:
I'm running E3V2-UBL-BLTouch-10x10-HS-v4.2.2-v2.0.1 with a CR touch and have not had any issues. 
I did find once that after many hours of printing, the bed seemed to lower some 10's of microns (or more likely the frame grew) and the first layer would no longer stick. I attributed it to heat propagation from the long runtime and built another mesh which I stored in address 1 instead of overwriting the current mesh in 0. Now I can switch between the two. I did change my bed to solid mounts (silicone standoffs) right after installing the CR Touch. I've been happy with the Jyers Firmware (and the CR-Touch). Thanks.

jpedlow, 2021-12-16:
On my 4.2.2 E3V2, running E3V2-UBL-BLTouch-10x10-v4.2.2-v2.0.1  -- my machine will only ever test 65 points, aka it says 65/100 then goes to complete. Interestingly the Mesh shows a 10x10 grid when viewing, but it didn't probe 100 points. I wonder if it's locked doing an 8x8 grid?

YClayton, 2021-12-16:
> On my 4.2.2 E3V2, running E3V2-UBL-BLTouch-10x10-v4.2.2-v2.0.1 -- my machine will only ever test 65 points, aka it says 65/100 then goes to complete. Interestingly the Mesh shows a 10x10 grid when viewing, but it didn't probe 100 points. I wonder if it's locked doing an 8x8 grid?

I never watched it run the whole 100 probes before. I just checked, and my E3v2 is also quitting after 65 probes. I wonder if it extrapolates the additional probe points. My probe also retracted two or three times, I didn't notice the probe count.

Here's some info I found over at the Marlin site (https://marlinfw.org/docs/features/unified_bed_leveling.html)...
_**Filling in the Mesh**
UBL includes a third phase, G29 P3, which fills in points on the mesh that were not probed automatically or manually. Note that unlike in bilinear leveling, UBL does not automatically extrapolate correction beyond the bounds of the mesh. If a mesh point is not defined no correction will be applied, and a missing point can affect up to 4 mesh cells._  The text continues if you'd like to go read it.

I suspect the firmware decides there are areas of the bed that are unreachable and fills in those points with extrapolated elevations; which would explain why it displays 100 points in the viewed mesh. In the end the created mesh does work well for me.
I've had - no issues - with the Z crashing into the bed.

jpedlow, 2021-12-16:
Perhaps - Maybe I'm expecting too much here from the firmware - my sense was, saying it's X/100 for probing positions is misleading if there's no expectation for the machine to be physically probing 100 points. But eh, fair enough, if it's averaging then all good.

YClayton, 2021-12-16:
> Perhaps - Maybe I'm expecting too much here from the firmware - my sense was, saying it's X/100 for probing positions is misleading if there's no expectation for the machine to be physically probing 100 points. But eh, fair enough, if it's averaging then all good.

I did some more research, and it seems probing 65 points when set to 10x10 is in fact normal operation. Upon completion the display states that - all reachable points - have been probed. This has to do with the offset of the probe from the nozzle.

It seems this has caused confusion multiple times and indeed an firmware tweak would be good. In the short term it could be stated for example. "Up To 10x10". I suspect the number of points to be probed can be calculated ahead of the probing operation. In which case, displaying...   X (1 to 65) of 65 points probed would be clearer instead of...  X of 100. Or a statement 10X10 selected - 65 points reachable. It's probably on the developers list of to-do items, but there are more pressing issues since it's working properly and just mis-reporting.

xcasxcursex, 2021-12-21:
Came looking for this same very specific thing. The probe is retracted and the nozzle crashed, on point 41, 10x10 HS

Patrixe, 2022-01-11:
Additional info: When the probing stops after point 65, I get the error "Unknown command: "5"" in my pronterface log.

nmkr, 2022-01-15:
can confirm crashing on point 42 with HS 10x10 UBL and stopping at 65 with you want to save eeprom, mesh looks ok

siahfri, 2022-02-04:
Same here. Happens every time I create a new mesh. Thought it might be a sick joke referencing HHGTTG

brdenos, 2022-04-06:
I am experiencing the same issue. Points 42, 84, 126. For the 15x15 HS firmware (actually Alex's firmware for Aquila, but same base code)