Jyers/Marlin #1647: [Feature Request] forcing to restore saved mesh, z offset, steps/mm, etc

dhupee, 2021-11-24:
### Description

After reading #1603 and just realizing that I actually relevel my printer after flashing every time without knowing I can just restore my setting is perhaps a "newbie mistake", and since many E3V2 users might use this firmware it would be handy to have an option to previously restored setting

I am a newbie to GitHub so this format might be wrong but correcting me would be helpful

thanks to Jyers and other contributors

tititopher68-dev, 2021-11-24:
@dhupee 
This is not the right location ... because it is not an issue ..
Sorry, but that does not make sense at the level of the firmware ... flashing a firmware, by its size or its content modifies the organization of the memory and therefore of the EEprom.  This implies that the data in the EEprom itself is corrupted.  That's why we insist on doing a Reset to Default, following by a Store Settings and following by a Restore Settings to initialize correctly the EEprom after flashing!

What you need to know about the organization of the EEprom.  The data is stored one after the other in a linear model, bits after bits.  A byte can store, for example, 8 Boolean variables or 2 integer variables or 4 Boolean variables and 1 integer variable, etc ... The size of the EEprom for the data saved by the firmware is 48 bytes (you can l  'worry-free expand to 64 bytes).  If you look at the dwin.h file (Marlin / src / lcd / e3v2 / jyersui), you will find a Class there called eeprom_settings.  The variables defined under this Class are in the form: 'type name: number of bits' ... we do not store the name of the variable.  During compilation, each variable is defined by its address in the EEprom and the number of bits.  Sorry, for being a little technical ...

Also how do you save data in EEprom before flashing firmware and recover it with corrupt EEprom ....
The only possibility is to go through Octoprint and a plugin allowing you to save data and then reintroduce it.

dhupee, 2021-11-25:
Thank you for your understanding to give me the explanation, cheer for the contributors

closed 2021-11-25 by dhupee