Jyers/Marlin #1651: Preheat mode incorrect

rzblue, 2021-11-28:
### Description

### Steps to Reproduce

1. Enter preheat menu
2. Select Bed or Hotend for Preheat Mode
3. Select a Preheat setting

**Expected behavior:** [What you expect to happen]
Selected device (Hotend or bed) should start to preheat

**Actual behavior:** [What actually happens]
Opposite device starts to preheat; ie. Bed preheats when Hotend is selected

#### Additional Information

I've modified `do_preheat` in `dwin.cpp` to the following:

```cpp
auto do_preheat = [](const uint8_t m) {
          thermalManager.cooldown();
          if (preheatmode == 0 || preheatmode == 2) { ui.preheat_hotend_and_fan(m); }
          if (preheatmode == 0 || preheatmode == 1) ui.preheat_bed(m);
        };
```
This has fixed the issue on my printer, however this doesn't quite make sense due to the contents of `preheat_modes` : `{ "Both", "Hotend", "Bed" }`

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

tititopher68-dev, 2021-11-28:
@rzblue 
Are your heating cables not inversed on the motherboard side ... Because I never had this issue during my tests ...

rzblue, 2021-11-28:
Nope, wires aren't reversed- this is the only thing that appears to be reversed (printing works fine)

Cal-LF, 2021-11-28:
I have found this to be accurate. Using V2.0.1 manual mesh. The preheats for bed and hotend are reversed.

tititopher68-dev, 2021-11-28:
@Cal-LF @rzblue 
It doesn't make sense to have to reverse the controls!  The code is correct, except that the {} are unnecessary ...
@rzblue 
By inverting, when you select Bed, does that trigger the heating of the Bed?

rzblue, 2021-11-28:
I'm aware it doesn't make sense. After changing the do_preheat function, the feature works as expected (selecting bed heats the bed)

Cal-LF, 2021-11-28:
@tititopher68-dev Yes the opposite preheat is selected and heated.

Cal-LF, 2021-11-28:
@tititopher68-dev @Jyers The preheat selections work properly in V1.3.4

tititopher68-dev, 2021-11-29:
@rzblue @Cal-LF @Jyers 
I integrated the various code modifications proposed by Marlin for the use of do_preheat(m) in the base of v1.3.4 ... and there, the preheat selection work properly.
@Jyers 
There are really too many bugs and inconsistent operations in the bugfix marlin base of the v2.0.1 release.

Ali229, 2021-12-11:
Just updated to latest, I'm having the same issue of temperatures being reversed.

mattcoon, 2022-01-04:
I can confirm the same behavior. Additionally, when using the preheat menu popup from change filament, selecting a filament type sets the preset for the bed, not the hot end, however, setting custom temperature works correctly.

DragRedSim, 2022-01-07:
Also confirming this issue on my unit.

I put together a custom build which confirmed the preheatmode variable is updating as expected (echoing to the display status line); I also note that the part cooling fan is only activating when both the nozzle and bed are set to heat (ie. preheatmode = 0). This suggests that the ui.preheat_hotend_and_fan() function is not working correctly.

A bit more digging, and I come across this in the apply_preheat() function: 
`if (TEST(pmask, PM_HOTEND))`
If we try to heat the hotend, preheatmode equals 1, calling ui.preheat_hotend_and_fan(m). This calls ui.preheat_hotend(m), which in turn calls ui.apply_preheat(m, PM_HOTEND); PM_HOTEND is now the value of our pmask. pmask is equal to 1.
But the TEST() macro becomes `(!!((n)&_BV(b)))`, which is bit-shifting the second value again... so if we ask for TEST against our pmask and PM_HOTEND, it returns a test against 1<<0x1, which is 2. !!1&2 returns false. No hotend heating. Then it tests against the PM_BED value; we get !!1&(1<<2), which becomes !!1&4, again false. Then PM_FAN tests !!1&(1<<4), !!1&16, obviously returns false. So nothing is dones. But now, ui.preheat_hotend_and_fan(m) then calls ui.preheat_set_fan(m), which calls ui.apply_preheat(m, PM_FAN). Our pmask is now the constant 1<<2, or 4. Nothing on the hotend, but we get to testing pmask against PM_BED... !!4&(1<<2), or !!4&4 - becomes !!4, or not(not(4)). not(4) is equal to -5, not(-5) is 4, and **anything that is not 0 is true**. So bed heating happens when we asked for the hotend (and by extension the fan).

Similarly, we ask for the bed to heat; preheatmode is 2. We call ui.preheat_bed(m); this is ui.apply_preheat(m, PM_BED); our pmask is 2. TEST(2, PM_HOTEND) is thus !!2&1<<1, which is !!2&2, which triggers a true result, and we get hotend heating! No bed or fan heating though.

In short, this is actually an upstream calculation bug, that was being worked around by the previous code ignoring these functions, and added in as part of the merging into mainline Marlin.

The correct bugfix is to change the function at Marlin/src/lcd/marlinui.h#L499, where the masks are defined, so that they are not bit-shifted before passing into tests, which will bit-shift them again. We can't change the macro itself as it is used in other locations in Marlin, and to do so would break substantial parts of it. Nowhere else in Marlin that I could find uses a bit-shifting within an enum to generate a mask.

Putting together some PRs now.

tititopher68-dev, 2022-01-07:
@DragRedSim 
I completely agree with this diagnosis.  But considering that Marlin produced this code for all UIs, I let them modify it ... Marlin sometimes declines to stop trying to do too much with the aesthetic side.  In version v2.0.2h (# 1694) I have reverted to something more pragmatic and it works.

DragRedSim, 2022-01-09:
Fixed upstream; until such time as Jyers returns and pulls from upstream, you can manually pull to your own forks if you wish, or apply the code patches I've made to your own files before compiling.

closed 2022-02-11 by Jyers

Ali229, 2022-03-27:
I'm getting this error while compiling after the merge:
Marlin\src\lcd\marlinui.cpp:151:49: error: 'PT_HOTEND' was not declared in this scope; did you mean 'PM_HOTEND'?