Jyers/Marlin #1656: random Z Axis "Falling" during idling

phenomeus, 2021-11-28:
### Description

I'm experiencing random Z Axis Stepper motor "disconnects" so the head is "falling" (**not crashing**) to the bed. 
its only happening during idling, preheat or between command moving Z axis manually, then homing, then for example create mesh. If I idle to long between this steps the z stepper is losing its "power"/ "grip" and the nozzle is traveling down, smoothly because of the build but never the less its loosing its Z position and the screen shows a Z position of "?" 

when I then go the menu to move/ raise the z axis I can feel how the stepper suddenly gets "power" to move and then it stays there until the next random fall

### Steps to Reproduce

I can't tell what is happening, just home the nozzle and wait? maybe 1 min, maybe 5 min

**Expected behavior:** as long as the machine has power the z stepper should hold the grip, position
i will find out if X and Y stepper are still enabled during this falling time.

**Actual behavior:** z stepper is losing/ getting disabled and the nozzle is "falling"

#### Additional Information

v2.0.1
Board 4.2.2
BLTouch, connected with normal port, not spliced

tititopher68-dev, 2021-11-28:
@phenomeus 
Not a firmware issue ! it's ever and only an hardware issue.
Please look at : https://github.com/Jyers/Marlin/discussions/1550#discussioncomment-1647754 and following link

phenomeus, 2021-11-28:
thank I will have a look

phenomeus, 2021-11-28:
are you sure its connected to the BLtouch drama? BLtouch is working flawless. I don't get it why the stepper motor just "forgets" to hold the power?! 

at this moment before "falling" there is no bltouch activity at all. it can be like 5 min between the two events

tititopher68-dev, 2021-11-28:
@phenomeus 
Yes, it is a problem known for a long time and often mentioned since the beginnings of the firmware but also since Smith3D.  There are random problems with the BLTouch port of the v422 and v427 motherboards (there is a hardware fix requiring to solder a 10 KOhm resistor as in the diagram of the related post).  This causes the Hotend to crash on the Bed.  That is why I recommend using the connection mode via the Z-Endstop.  But that requires modifying the configuration.h file and recompiling everything.
 Personally, I consider v2.0.1 too unstable and this comes from the Marlin code ... as moreover Marlin intervenes directly on the Jyers code in the bugfixes and sometimes by creating other bugs as with the preheats ... I follow  stay on the basis of the v1.3.4 upgrade with my own features and those of the v1.3.4 and all is well ...

Cal-LF, 2021-11-28:
@phenomeus @tititopher68-dev V1.3.4 is the stable version I use but have been testing with latest version. There are many bugs to be addressed.

tititopher68-dev, 2021-11-28:
@phenomeus 
Oops <I'm experiencing random Z Axis Stepper motor "disconnects" so the head is "falling" (not crashing) to the bed.>

Ok also, check the geometry of the printer, the vSlot for the Z axis et above all the position of the Z-axis screw, the trapezoidal screw and your coupler. The slightest angle shift of the trapezoidal screw and it will go down on its own.

phenomeus, 2021-11-28:
> When the motors are not activated within 120 seconds, Marlin deactivates them (at this point, you should see the position of the axes on the screen flashing).
> 
> Ok also, check the geometry of the printer, the vSlot for the Z axis et above all the position of the Z-axis screw, the trapezoidal screw and your coupler. The slightest angle shift of the trapezoidal screw and it will go down on its own.

that sounds more like it. so I will measure the time, maybe its not that random. should the axis stick in the z position,  even when the steppers are disabled? I thought its a sign of a better build when the z is sliding easily downwards after disabled steppers.

tititopher68-dev, 2021-11-29:
@phenomeus 
Even when the steppers are disabled, the Z-axis should stick in position.
The axis can sometimes move very very slightly, but because of the clearance between the brass nut on the X axis support and the trapezoidal screw, which an anti-backslash corrects.

phenomeus, 2021-11-30:
i have a pretty heavy z axis, but never the less I made some maintenance and adjusted the nuts so now, when I go up 100 and disable the stepper, its falling just about one nut rotation where its getting catches and hold. 

maybe the is a good video or something how It can be setup perfect.

tititopher68-dev, 2021-12-01:
@phenomeus 
If your X axis is too heavy then installing a Dual Z may be a good idea.
[image]
[image]

Or this : 

[image]

phenomeus, 2021-12-01:
what are the also benefits of a dual Z setup? its again an invest to make

tititopher68-dev, 2021-12-01:
@phenomeus 
The general concept is to provide additional stability during the print. You could run the risk of the -X- axis sagging due to the weight of the print head and/or additional wear on the rod or bearings on the one side (smooth side) as a result of the off balance weight.

Namely, the more weight there is on the X axis, the more the motor torque required in Z will vary depending on the position of the head.
Take a look at : 
https://youtu.be/6UgT9YqY3UA

In your case, you could always install an anti-backslash, this will limit the problem.  A dual Z with a single engine is also a solution with additional advantages.
The cost is not that high (€ 11.99 | 3D Printer Accessories, Dual Z Axis Upgrade Kit, Dual Z Tension Pulley Assembly for Creality ender-3v2, Dual Z Axis Upgrade Kit
 https://a.aliexpress.com/_mt4HK8K).

phenomeus, 2021-12-01:
I watched the video and I like the idea to upgrade step by step from 1 to maybe even 3

mikeoverbay, 2022-03-05:
The easiest and cheapest way to stop the Z from dropping when power is removed is to get an [anti-backlash nut.](https://www.amazon.com/Backlash-Elimination-Upgrade-Tornado-Threaded/dp/B08LZ1V56T)