Jyers/Marlin #1657: missing main print screen/ scrollable menu after filament change during print

phenomeus, 2021-11-28:
### Description

when you are printing and use the "change filament" option, the print resume but you can't get out of the menu to the main menu anymore until the print is finished

### Steps to Reproduce

1. print something
2. use change filament
3. resume print even without changing the filament
4. wait and try to scroll when print is resumed
5. no scrolling so no back button to reach 
6. so no main menu anymore

**Expected behavior:** scrollable menu stays so I can reach "back" (button at the top)

**Actual behavior:** no scroll function anymore, I have to wait until print is finished, I even can't use the "tune" menu anymore

#### Additional Information

4.2.2
v2.0.1
e3v2+bltouch

Cal-LF, 2021-11-28:
Using the latest version for manual mesh, I am able to use the filament change option in the tune menu. It returns to the tune menu and there is no hilighted selection until the knob is rotated. I got some menu artifacts displayed along with the hilighted selection and was able to select anything in the tune menu.
There may be a bug not allowing the scrolling to return but I tried it twice and functioned normally.

memomedfish, 2021-12-02:
I have the same problem, it needs to be addressed.

hakancez, 2021-12-15:
Same problem here.