Jyers/Marlin #1674: Bltouch raster and hs mode menu item possible ?

Bersaker, 2021-12-06:
Can you create a menu item in which you can choose the grid yourself, whether 5x5 10x10 or 15x15 can be selected via bltouch? And a menu item to change hs Mode?

tititopher68-dev, 2021-12-06:
@Bersaker 
These features are not implemented in the last bugfix_marlin and I can almost advocate that @jyers do not implete them so Marlin does not do it ...

I think you make references to Mriscoc? Ah, it's sure that with him, you will never have the source code for this BLTOUCH functions. He does not respect the Marlin license, forcing him to publish the source code of the releases on line (on his Github, the sources are absolutely not those of his releases, but that's another subject ... Advice, do not insist on asking where you will be blocked and banned!). If you want these functions, use his firmware (with some bugs ...) But you will be forced to do with him for bugs and you will not have the choice to customize your own configuration!

rzblue, 2021-12-15:
With regards to the technical aspect of those features, currently the grid size and hs_mode are configured with pre processor macros and hardcoded at compile time to reduce program size. It would require a massive restructuring of the bltouch code to be able to make those changes at runtime.

tititopher68-dev, 2021-12-15:
@rzblue 
Not quite correct, #define macros can be associated with one or more variables, a function, etc ... This is how 4 corners Leveling works:
`#define PROBE_X_MIN _MAX (0 + corner_pos, X_MIN_POS + probe.offset.x, X_MIN_POS + PROBING_MARGIN) - probe.offset.x`
During compilation PROBE_X_MIN is defined only once, but is not fixed and will depend on the values ​​of the corner_pos and probe.offset.x variables which can be modified in the UI.

So for GRID_X_MAX / GRID_Y_MAX, it is easy to do and the size of the code is hardly affected.

But for HS Mode, it is much more complex because it is a simple #define without variable (#define BLTOUCH_HS_MODE), there it is to associate a boolean variable, to modify #define BLTOUCH_HS_MODE in #define BLTOUCH_HS_MODE true and act on all the files where this macro is referenced ... i.e. 13 files (Configuration_adv.h, bltouch.h, bltouch.cpp, G34_M422.cpp, G35.cpp, language_en.h, M401_M402.cpp, menu_bed_corners .cpp, menu_configuration.cpp, menu_tramming.cpp, motion.cpp, probe.cpp and settings.cpp). The size of the code is also little impacted (for having done it ...), because in fact we reduce the 2 cases in 1 only.

But I don't think @Jyers will implement it until Marlin does .... and right now it's nada ...