Jyers/Marlin #1676: Extremely fast extruder speed if not homed

crispyjones, 2021-12-07:
### Description

If printer is turned on and a manual move is attempted from the menu Prepare>Move>Extruder {any amount} **before** homing, the extruder moves extremely fast causing filament grinding and jumping. Once the printer is homed the extruder speed is normal and e-steps can be calibrated normally. The X, Y, Z and extruder speeds are all default on my E3v2.

### Steps to Reproduce

1. turn on printer (do not home)
2. set hot end temp (do not home)
3. manually extrude any amount from menu (beware anything over 30-40mm will likely grind) and observe fast extruder speed
4. home printer
5. repeat 3 and observe normal extruder speed

**Expected behavior:**
Extruder will run at the user set speed and e-steps can be calibrated normally.

**Actual behavior:** 
Extruder runs **very** quickly, causing terrible noises and grinding and the inability to calibrate e-steps until printer is homed.

xxqpxx-pmw, 2022-01-03:
Same. Thank you crispyjones, for figuring this out. I was about to reflash stock fw until I checked here and saw your post.

Ali229, 2022-01-30:
Same issue

sauroweb, 2022-03-15:
Same issue. Motherboard 4.2.2.

Ali229, 2022-03-27:
Any update on this?

aleite93, 2022-04-10:
Same issue