Jyers/Marlin #1679: Cancel print crashed nozzle into printed object

eloo, 2021-12-09:
### Description

Hi, when i cancel a print on the display the nozzle will everytime try to go Z25.

If the print is already higher than 25mm it crashed into the print. 

### Steps to Reproduce

1. Print something higher than 25mm
2. Cancel the print on the display
3. See the nozzle going down and crashing into your print

**Expected behavior:**

I would expect the nozzle to liften a bit or at least to not change its Z position.

**Actual behavior:** 

Nozzle tries to go to Z position 25 

#### Additional Information

Version: Jyers-2.0.1-E3V2-UBL-BLTouch-10x10-v4.2.2-v2.0.1.bin

tititopher68-dev, 2021-12-09:
Not a firmware issue. Check your endGcode in your slicer. You should have a Z axis move command running in absolute value instead of relative value ... so you should add before a G91.  And in this case, it does not matter whether it is during printing or at the end of printing, after a Stop Print, your axis should lift 25mm from its last position.

eloo, 2021-12-10:
Okay, my end g-code in cura looks like this:

```
G91 ;Relative positioning
G1 E-2 F2700 ;Retract a bit
G1 E-2 Z0.2 F2400 ;Retract and raise Z
G1 X5 Y5 F3000 ;Wipe out
G1 Z10 ;Raise Z more
G90 ;Absolute positioning

G1 X0 Y{machine_depth} ;Present print
M106 S0 ;Turn-off fan
M104 S0 ;Turn-off hotend
M140 S0 ;Turn-off bed

M84 X Y E ;Disable all steppers but Z
``` 

for me this looks like the z should be raised and not lowered. 
also rest is of this gcode seems not be executed when i abort.. because the bed is also not moving forward and so on.

tititopher68-dev, 2021-12-11:
@eloo 
Nothing abnormal indeed in your endGcode.  You don't print from Octoprint by chance because Octoprint uses its own startGcode and its own endGcode in case of cancel, likewise it has its own endGcode at the end of printing?

I also strongly advise you to apply this fix for maple -> timers and pwm HAL :
Just copy the 2 files timers.h and fast_pwm.cpp in Marlin/src/HAL/STM32F1
These Files are compatible with Jyers Releases from v1.3.4 to actual v2.0.1
This should prevent some Hotend crash on the Bed, some issues with Endstop, BLTouch and LCD Glitch...
[Timers_and_Maple_pwm_fix2.zip](https://github.com/Jyers/Marlin/files/7652446/Timers_and_Maple_pwm_fix2.zip)

I had forgotten to clarify that it was necessary to modify in the Hal.h (Marlin / SRC / HAL / STM32F1) file:

`#define HAL_CAN_SET_PWM_FREQ      // This HAL supports PWM Frequency adjustment`

by : 

```
#ifndef PWM_FREQUENCY
  #define PWM_FREQUENCY      1000 // Default PWM Frequency
#endif
#define HAL_CAN_SET_PWM_FREQ      // This HAL supports PWM Frequency adjustment
```
You will then need to compile your own firmware.

C-Noob404, 2021-12-11:
I had exactly the same problem with the Ender 3 v2. Crashed my nozzle into the bed. Went straight back to v1.3.5.b and everything worked just fine. Pls check again.

eloo, 2021-12-12:
@tititopher68-dev 

good hint with Octoprint. 
Yes i'm using octoprint and i guess i have found the issue.

I had the Z+25 in my octoprint and was assuming that this is relativ to the current position. 
But i guess its not relativ but absolut. So it was 0 + 25.. 

I have now adjusted my code in Octoprint to 
```
; disable motors
M84

; disable all heaters
{% snippet 'disable_hotends' %}
{% snippet 'disable_bed' %}
;disable fan
M106 S0

; relative XYZE
G91
M83

; retract filament, move Z slightly upwards
G1 E-1 F1000
G1 Z+5 F4500

; absolute XYZE
M82
G90

; move to a safe rest position, adjust as necessary
G1 X0 Y200
```

But as @C-Noob404 also mentioned.. maybe there was a change from 1.3.5.b to 2.x

Because in the old version i could not remember to have this issue with my old code.
Maybe the Stop/Cancel GCODE in the firmware has changed?

tititopher68-dev, 2021-12-12:
@eloo
I differentiate the code of the UI and that of the Marlin base.  The bugs come from this currently very unstable base.  Marlin has taken a direction that will penalize many printers and motherboard manufacturers at least for a potentially long time.  (Note that the BTT SKRs also have these same problems. Worse still because Marlin does not want to support the management of the GD32 and N32 CPUs (which equip the Aquila and the SKR BTT E3 mini ...). Indeed, STM32  could attack them for lack of license! I find that pitiful on the part of STM ...).

I don't think @ C-Noob404 is having the exact same problem.  There have always been Hotend crashes on the Bed related in part to the new handling of the STM32 CPU, which got worse long after the Marlin base of v1.3.5b.

The crashes can be the consequence of 2 elements, a design defect of the motherboards in the 5v power supply of the port for the BLTouch and / or a bad management of the timers and the PWM of the CPU from where data collisions for example.  for motor steps or reading via BLTouch.  The fix I propose solves this problem (but not the 5v one where a hardware modification is necessary).  In your case, it is not a crash but a question of relative / absolute distance control.
 
However not all bugs in the v2.0.1 Marlin base are fixed including by merging the latest Marlin Bugfixes ... which worries me, this branch has never been as unstable as it is now.

tititopher68-dev, 2021-12-16:
@eloo @C-Noob404 
Try this new release (that I built) for the Jyers's Firmware with maple fix, Gcode Preview and Reset Counter : Marlin-bugfix-JyersUI -v202 (with the latest Marlin bugfix too) :
[Marlin-bugfix-JyersUI-v202.zip](https://github.com/Jyers/Marlin/files/7731013/Marlin-bugfix-JyersUI-v202.zip)