Jyers/Marlin #1684: [BUG] in the latest Jyers firmware Layer height starts at 0 not 1

Geo112178, 2021-12-13:
Hi i am running the latest  firmware of Jyers E3V2-UBL-BLTouch-10x10-v4.2.2-v2.0.1.bin and i fount a bug i was printing with 2 colors and i told Cura to stop printing on layer 41 and it stopped at 40 and didnt understand why i thought i put the number in wrong but i didnt so i just hit continue and it was doing what the other color was supposed to do so i just stopped it. Today i was starting another print I realized that the Layer height started at 0 and not 1 i have it displayed on my screen. Now i know why from before why it stopped at 40 instead of 41 there must be a bug so please fix this issue thank you.  

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

tititopher68-dev, 2021-12-13:
@Geo112178 
????
Absolutely not a firmware issue !
It is not the printer that determines the layer number but your slicer only.

Geo112178, 2021-12-13:
> @Geo112178 ???? Absolutely not a firmware issue ! It is not the printer that determines the layer number but your slicer only.
but the previous jyers never gave me this problem

tititopher68-dev, 2021-12-13:
@Geo112178 
Github sent me your response by email which you must have deleted afterwards.
Let us consider Cura (4.10) for example, if you open the produced gcode with a text editor, you can observe that it is indeed the slicer which numbers the layers and that it starts with layer 0 (1st printed layer) and not 1... It remains to define, according to your choice, whether the operation should be done before the layer change or after...

Example (Cura) : 
```
; Ender 3 Custom Start G-code
G92 E0 ; Reset Extruder
M280 P0 S160 ; BLTouch alarm release
G4 P100 ; delay for BLTouch
G28 ; home
M420 S1
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish
G92 E0
G92 E0
G1 F2700 E-5
;LAYER_COUNT:260
**;LAYER:0**
M107
G0 F6000 X108.673 Y69.044 Z0.2
;TYPE:SKIRT
G1 F2700 E0
G3 X111.339 Y66.194 I7.684 J4.516 E0.13670 F1500
G3 X115.427 Y64.680 I5.373 J8.231 E0.28953

```

Geo112178, 2021-12-14:
i was my fault i figured it out thank you