Jyers/Marlin #1685: X Axis crash after flash

ase1590, 2021-12-14:
Printer: Ender 3 V2 (Purchased ~1 week ago) w/ 4.2.2 board

Precompiled Firmware: E3V2-BLTouch-5x5-v4.2.2-v2.0.1.bin

### Description

The X Axis endstop switch seems to no longer register after attempting to level the bed before doing a "home All".

After leveling the bed, the printer will instead move the nozzle to the right slightly away from the X axis and stop when attempting to home the X axis specifically. 

Y axis and the BlTouch Z axis work fine. 

### Steps to Reproduce

1. have hotend at home position
2. Flash new firmware
3. forget to clear EEPROM
4. skip homing the bed
5. immediately try to use the 'level' option
6. after completion, attempt to home each axis independently
7. X axis no longer functions

I've verified with a voltmeter that my endstop switch works as expected. I had this issue once earlier after coming straight from the stock creality firmware, but Re-flashing at the time seemed to clear whatever strange bit was set, and then the X axis worked normally. 

After installing my BlTouch however, Reflashing this firmware or the stock creality firmware seems to have the issue persist. 

running m502 & m500 gcode commands have not fixed this issue either. Not sure whats going on here.

tititopher68-dev, 2021-12-14:
@ase1590 
This cannot be a firmware issue since you have this problem with the BLTouch regardless of the firmware.  I am thinking of a purely hardware problem.  Check the cable of the X endstop.  How is your BLTouch connected?
On the other hand, it is imperative after flashing a firmware to do a Reset to Defaults following by a Store Settings and following by a Restore Settings to initialize correctly the Eeprom and all's parameters.

ase1590, 2021-12-14:
This has to be at least partly a firmware issue, as the issue does not exist on stock 1.0.2 Creality firmware, which is based on Marlin 2.0.1

I do however see this behavior in the later creality stock version 1.1.2 as well as in Jyers firmware.

I will have to do some testing to see if I can narrow down a change that forces this to occur or if it is related to adding support for BL Touch that causes the end stop issues.

It is strange to me that the stock old firmware without Bl Touch work fine, but later versions with Bl Touch support do not.

ase1590, 2021-12-14:
Alright, so using the 3x3 Bl Touch Jyers firmware correctly handles the X axis endstop. This may be a problem then specifically with the 5x5 pre-built firmware.

tititopher68-dev, 2021-12-14:
@ase1590 
The stock firmware version 1.0.2 is based on a very old version of Marlin (more than 2 years).  The version 1.1.2 is based on Marlin v2.0.6 ... and the version of Jyers's firmware is based on Marlin Bugfix v2.0.9.2 .... The code for the management of X or Y endstops comes only from Marlin and is there was a problem, it would have been a long time since this was reported and fixed.  The pre-compiled BLT 3x3 or 5x5 versions are identical except for the definition of the number of measurement points, i.e. the only parameter #define GRID_X_MAX in configuration.h .... nothing to do with endstops near or far.

According to  my opinion, your problem is hardware related to the BLTouch, how and why, I can not say.  Maybe a problem of mass, of interference.  All I know is the BLTouch port is a problem on some motherboards.  Try twisting the cable of the X Endstops and the cable of the BLTouch, this may neutralize some interference. 

Then I also strongly advise you to apply this fix for maple -> timers and pwm HAL :
Just copy the 2 files timers.h and fast_pwm.cpp in Marlin/src/HAL/STM32F1
These files are compatible with Jyers Releases from v1.3.4 to actual v2.0.1
This should prevent some Hotend crash on the Bed, some issues with Z-Endstop (via port of BLTouch) and LCD Glitch, steppers ...
[Timers_and_Maple_pwm_fix2.zip](https://github.com/Jyers/Marlin/files/7652446/Timers_and_Maple_pwm_fix2.zip)

I had forgotten to clarify that it was necessary to modify in the Hal.h (Marlin / SRC / HAL / STM32F1) file:

`#define HAL_CAN_SET_PWM_FREQ      // This HAL supports PWM Frequency adjustment`

by : 

```
#ifndef PWM_FREQUENCY
  #define PWM_FREQUENCY      1000 // Default PWM Frequency
#endif
#define HAL_CAN_SET_PWM_FREQ      // This HAL supports PWM Frequency adjustment
```
You will then need to compile your own firmware.

Learning to compile your own firmware has become necessary, and it is not difficult.  There are mini tutorials in the Discussion area.

ase1590, 2021-12-14:
I'll note down the PWM fix for later, and compile that in once I figure out this end-stop issue.  Haven't had the hot-end crash into the bed during leveling just yet, but better to be safe than sorry. 

I think you're correct on this being a hardware bug. The 3x3 firmware that this was working on suddenly stopped working after a bit. Flashing back to the old firmware did not fix it this time. The switch itself is fine. The problem must lie either in an intermittent wire connection or perhaps an issue with the PCB itself. I will have to track this down. 

I am going to close this for now, as it seems your original assumption of this being hardware related issue is most likely correct. It just happens to be the 'fun' intermittent kind of hardware issue....

Thank you for the info on suggesting the pwm fix, I've added that to my documentation for later :smile:

closed 2021-12-14 by ase1590

Mineplayerminer, 2021-12-14:
Weird, I tried to reproduce the problem. I think the EEPROM has cleared itself after I upgraded my Firmware from ManualMesh to BLTouch. Mine was fine.

ase1590, 2021-12-15:
Updating this.

Pulled out the X axis connector to themainboard to find that the end stop internal metal crimp had a hairline crack and was hanging on by a thread of metal. Ugh.

At least I know now where the problem lies

ase1590, 2021-12-15:
@tititopher68-dev do you have some info on what the Timers/PWM patch is attempting to do? I'm not super familiar with the STM32 platform or marlin firmware just yet.

tititopher68-dev, 2021-12-16:
@ase1590 @Mineplayerminer @Jyers 
Try this new release (that I built) for the Jyers's Firmware with maple fix, Gcode Preview and Reset Counter : Marlin-bugfix-JyersUI -v202 (with the latest Marlin bugfix too) :
[Marlin-bugfix-JyersUI-v202.zip](https://github.com/Jyers/Marlin/files/7731013/Marlin-bugfix-JyersUI-v202.zip)