Jyers/Marlin #1691: "Display off" feature when using OctoPrint

wekii, 2021-12-16:
### Description

Is there any way to trigger "Display off" command by using OctoPrint?

tititopher68-dev, 2021-12-16:
@wekii 
Not an issue ! Please look at the  list of Gcodes Marlin on the web...
M251 B0 -> Off
M251 B255 -> On

wekii, 2021-12-16:
Thank you so much. It works. I searched everything and tried to google many keyword combinations but without success.
You saved me.
Thanks a lot.

dev-inciter, 2022-07-25:
Hey wekii,

the user that answered your question seems to be deleted and it's posts with him. Could you please tell me how to turn off the display?

Thanks

wekii, 2022-07-25:
M251 B0 -> Off
M251 B255 -> On
I hope it helps you, let me know.

NoQuarter89, 2022-07-27:
> M251 B0 -> Off M251 B255 -> On I hope it helps you, let me know.

Hey wekil! Today I updated to the latest release for the first time in almost a year. The m251 codes don't seem to work anymore . Have they changed? I compiled from the main bugfix branch with bltouch 3x3 as my config template.

wekii, 2022-07-27:
Hi,
today I finally found the entire c/p of the answer that I got here:

Since 2.0.1 :
M256 B0 -> Off
M256 B255 -> On

For v 1.3.5b and old :
M251 B0 -> Off
M251 B255 -> On

Hope it helps, let me know.

NoQuarter89, 2022-07-27:
Amazing! Thank you very much.

dev-inciter, 2022-07-28:
Yeah, using the brightness value is clever. That did the trick, thank you!