Jyers/Marlin #1706: Continuous Beep from board when bed is heating

BoredFishRE, 2021-12-18:
### Description

When the bed is heating, a continuous beep is produced from the board. Once the bed reaches temp, the beep stops. 

Ender 3 V2 with CR Touch, 4.2.2 BLTouch 3x3 firmware.

### Steps to Reproduce

1. Load GCode
2. Start print.

**Expected behavior:** [What you expect to happen]
The bed just heats.

**Actual behavior:** [What actually happens]
While bed heats, there is a continuous beep.

#### Additional Information
Video here: https://drive.google.com/file/d/1TwHMNWu42UjvQOBb27WzDzT3VwBR5DU8/view?usp=sharing

tititopher68-dev, 2021-12-18:
@BoredFishRE 
Sorry , but absolutely not a firmware issue...certainly an hardware issue. Check the cable and the connector for the Bed. The bed resistance must be between 2,5 kOhms and 2,8 kOhms. Check too the connector on the motherboard. Check the ground (power).
Are you using an UPS ?

BoredFishRE, 2021-12-19:
> @BoredFishRE 
> Sorry , but absolutely not a firmware issue...certainly an hardware issue. Check the cable and the connector for the Bed. The bed resistance must be between 2,5 kOhms and 2,8 kOhms. Check too the connector on the motherboard. Check the ground (power).
> Are you using an UPS ?
> 
> Note : I advise you to compile your own firmware with these updated sources : https://github.com/Jyers/Marlin/discussions/1694#discussion-3749966

I am running off a UPS, I'll move it to just surge protection and not battery power. It's unusual though, because normally when it's drawing too much power, my UPS screams at me.

BoredFishRE, 2021-12-19:
Yep. Looks like it was the UPS. Thanks!

closed 2021-12-19 by BoredFishRE

Mineplayerminer, 2021-12-19:
It just draws too much current from the UPS and the UPS tries to warn the user of the overcurrent.