Jyers/Marlin #1709: Just updated to Jyers ender 3v2

kshaub103, 2021-12-20:
Just updated and now I'm not sure what I did wrong just install this linear rails and it was working fine before the update. Here a video. When I home this is what happened. I'm running ender 3v2 direct
Drive linear rails xyz octoprint jyers Firmware 

https://user-images.githubusercontent.com/96362402/146707707-f97f1cec-2ebd-452c-b027-9fcc2b391aaf.MOV

kshaub103, 2021-12-20:
Restart just did the same thing btw.

tititopher68-dev, 2021-12-20:
@kshaub103 
It's certainly an issue with the HAL CPU code by Marlin (steps motors issue, BLTouch issue, Endstops issue, etc...).
I advise you to compile your own firmware with this sources : https://github.com/Jyers/Marlin/discussions/1694#discussion-3749966
Don't forget after every flash to do a Reset to Default, following by a Store Settings and following by a Restore settings to initialize correctly the EEprom.

kshaub103, 2021-12-20:
I'm new to this firmware is there a walk thru or instructions?

kshaub103, 2021-12-25:
Updated didn't seam to help now the printer just keeps erroring out. Anyone able to help the IO seams great but it useless if I can't get my printer to work with it. Anyone

tititopher68-dev, 2021-12-25:
@kshaub103 
Sorry but the v2.0.2.g works well. It's some thing now with your setup but not the firmware...

kshaub103, 2021-12-26:
I'm glad it works for you and I'm sure a bunch of other. I have hope but I'm just not seeing what is wrong. Using the pre configuration coded 2.02 5x5 with BL touch for the 4.2.2 board.

tititopher68-dev, 2021-12-26:
A user had the same issue overnight as on your video and it was a failed Endstops switch.  Your problem is hardware and it is always much more complex to pinpoint.

kshaub103, 2021-12-26:
https://user-images.githubusercontent.com/96362402/147414796-dc752381-a919-485e-98be-ed19ed260005.MOV still need help error

kshaub103, 2021-12-26:
Homing failed reset

kshaub103, 2021-12-26:
Home all 

https://user-images.githubusercontent.com/96362402/147415684-ec7249fe-62c0-4fda-a5f8-728305bbec13.MOV

tititopher68-dev, 2021-12-26:
@kshaub103 
A failed homing is usually caused by Endstops.  Your BLTouch also behaves like an Endstop.  It seems to be faulty, because it went into alarm (according to your video) ... At worst your motherboard is faulty.  A power supply fault, in particular a grounding fault, can generate interference ... and this kind of problem.  I would check the masses because it seems to affect all the contactors (which work by groundings).

kshaub103, 2021-12-26:
So what do I test? How to test main board or end-stops

kshaub103, 2021-12-27:
So what do I do now?

kshaub103, 2021-12-27:
What mainboard should I update to? What kinda of new end stops? Which mean well power supply. Maybe a new printer?

tititopher68-dev, 2021-12-27:
@kshaub103 
To test endstops switch: use a multimeter and test the continuity with contact open then contact closed. (same for cables).

To test the ground, when you look at the motherboard, at the bottom left you have 3 pairs of green connectors. Consider the pair (Nozzle wire port) right next to the white connector (Filament sensor port) and disconnect the right wire on this green connector.

Then test the heating of the hotend. If the mass is correct, you should have a Printer Kill after 30s to 60s. Ditto for the Bed: Reconnect the previous wire and disconnect the right wire of the second pair from the green connector (Hotbed wire port).

For my part, I modified as for the ender2 pro, the grounding by adding 1 spacer screw et 2 ground wires between the socket and the motherboard and between the power supply box and the motherboard:

[image]

kshaub103, 2021-12-27:
Thanks I will try asap I do really appreciate your assistance and my annoying question. I'm hoping to get this corrected. Planning on just upgrading to another board any suggestions?

tititopher68-dev, 2021-12-27:
@kshaub103
I don't know if you have had your printer for a long time. It is interesting to do a check up every 100h-200h of impressions.
I don't think your motherboard is out of order. Check all the screw connections, the plugs ... A good grounding is not a luxury, rather an advantage (4 terminals and 2 ends of wire).

On Amazon, you'll find Creality Switch Endstops for cheap. For example a kit 3 endstops + 3 thermistors + 3 heating cartridges ... for 12 € / $.

Creality mainly sells MB v4.2.7. Soon the MB v4.2.3 will be available. I have an ender2 Pro which was sent to me for testing and which has this MB. It corrects the problem of the dedicated BLTouch port and I have the impression that there is a connection for a WiFi adapter. I can use the firmware with the screen of the Ender3V2 ... A kind of Prusa mini!

The BTT SKR E3 mini v2 motherboard (not the v3, still some problems with Marlin). It requires an adaptation of the cable for the screen but it offers the Linear Advance, more expensive on the other hand and especially some arrives with a GD32 instead of an STM32 with less RAM (48K instead of 64K) and flash memory reduced to 256K instead of 512K .... Of course with the idea of ​​keeping this firmware.

In terms of BLTouch (not CRTouch, catastrophic, too many problems reported), Antclabs v3.1 (generally those from Creality), 3Dtouch v3.0 from Unitak or TriangleLab work well, but other Geetechs, etc...not always recognized or well managed.

mrrstrat, 2021-12-29:
I agree with @tititopher68-dev - check connections is in order. Complete reset. I am using the current release of the firmware I have compiled on my Ender 3 V2 with CR Touch: 5x5 mesh, BL Touch features enabled. All the stock stuff worked first time every time for me. The issue originally reported looks like a lost home state - end stops not working perhaps.

The end stops are cheap and fragile but they are very inexpensive. I bought 10 of them to keep on hand as I broke the metal tab off my Y axis home switch when it snagged on my shirt. I got them on Amazon and its a nice part to have plenty of them.

Testing them? if the little "clicker" is working then in my experience with micro switches it is fine. Testing is a breeze with a meter - most new meters have a beep continuity check - these switches are normally closed and clicking it should make meter beep when the two probes touch the contact pins.

Embedded device programing is all about state, changes in state, and checking for state so a lost initial state might be an init problem or re-init of environment from reentrant code. The physical machine state needs to be properly reset - EPROM reset process described above by @tititopher68-dev. The machine will keep settings state with power off - I assumed a re-flash re-inits all of this but never tested that: evidently this may not be the case and requires manual clearing. With a new firmware flash the initial state might be compromised if the physical hardware is not completely reset.

method1020, 2022-01-01:
I am currently using 2.0.1 release on a 4.2.2 Ender 3 v2. No current issues here.

kshaub103, 2022-01-01:
I'm glad that it seams to be an isolated issue. @method1020

kshaub103, 2022-01-03:
Installed new mainboard still waiting on new end stops. Didn't see any unconnected or any ground wires. I have seen the post on the motor being grounded but any one having a suggestion on what to ground too? I like @tititopher68-dev idea for his ender 2. Hoping to have the printer back online and resolve this issue final.

kshaub103, 2022-01-03:
https://user-images.githubusercontent.com/96362402/147894272-0b27d89a-c2c2-4fe0-b6ee-32853e92a769.MOV

When to move the printer back in to the enclosure I build and saw my bl touch seamed to light up with out the power. Do I need to cover the bl touch wires or what? I'm if the linear rails have something to do with it but I'm not sure I want to proceed with power until this is revolved.

mrrstrat, 2022-01-03:
Strange and very twilight zone. Grounding is important on the motherboard - the screws seem to be all that's needed. The servos are touching metal frame but I do not believe the frame is used at all for the servos. There simply is no provision from what I can tell that chassis grounding is important to the servos. The Mean Well 350-24 series seems to have sufficient circuit  isolation from the 24v B+ delivery that tells me catastrophic failure looks to be handled sufficiently prior to B+ output to the HE, bed, and mother board. And the 24v devices (servos) electrically are happy without chassis grounding.

mrrstrat, 2022-01-03:
Check your physical plug and cable connections if you did a board swap - it is possible to mis-plug or have an issue not easily able to be seen between board and devices plugged into it.

tititopher68-dev, 2022-01-03:
@kshaub103 @mrrstrat 
It's just physical science.  Nothing to do with the ground in this case (see the last video).  A motor turns into a current generator.  When it is powered, the stator is actually powered, which causes the rotation of the rotor and therefore the displacement of an axis.  (I skip over the details of what's going on between the stator and the rotor).  But if the power supply is removed and the axes are moved, the rotor is driven in the stator, which under the effect of a variation in the magnetic field induces an electric current which can be sufficient to provide sufficient energy for  power the motherboard and peripheral like the bltouch.  And faster you run the motor, more current you induce.  Some poorly protected motherboards can even burn out!  Also never move an axis by hand very quickly!

mrrstrat, 2022-01-03:
Yes I am aware of the effect: it is how A/C power is made :-)

Such an effect is likely to be normally factored into the design of the system - most mechatronics designers know about this and other effects on the system. I would think damage is theoretically possible but occurs infrequently (or even rarely).

kshaub103, 2022-01-03:
So I should be good to go? Makes sense that the motor created the charger I was moving it and the bed moved forward so I figured I would film the effect. Hoping I didn't damage anything

mrrstrat, 2022-01-03:
I @tititopher68-dev and I suggest - it is theoretically possible, but I am thinking highly unlikely. If I as a design engineer designed something that moving one of the parts induced catastrophic failure then the shame on me lol. But follow best practices in any hardware changes as even a good design cannot protect you against improper usage.

closed 2022-01-29 by kshaub103