Jyers/Marlin #1710: Software Endstops Menu not displayed

benebrady, 2021-12-20:
### Description

Software Endstops Menu not displayed on DWIN screen when the option is turned on in Configuration.h

### Steps to Reproduce

1. Turn on Software End stops menu in Configuratiobn.h
2. The software end stops menu is normally displayed in the Marlin stock Creality display under the Move Axis submenu.
3. The software end stops menu is normally used to turn OFF or ON the software end stops when setting the Z offset for the BLTouch. 
4. The setting of the Z offset cannot be accomplishes as the default setting for the Software End Stops is ON and it can't be toggled to OFF to allow the movement of the nozzle to negative values as required when setting the Z offset value.

**Expected behavior:** [What you expect to happen]
the display of the menu as needed.

**Actual behavior:** [What actually happens]
No menu option is displayed to toggle the Software End Stops

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.
[Ben E. Brady Configuration - BLTouch and runout.zip](https://github.com/Jyers/Marlin/files/7748360/Ben.E.Brady.Configuration.-.BLTouch.and.runout.zip)

tititopher68-dev, 2021-12-21:
@benebrady 
No, sorry but you are wrong.  No issue!  There has never been an Endstops menu unless you chose the DWIN_CREALITY_LCD_ENHANCED.  Here it is DWIN_CREALITY_LCD_JYERSUI.  It is true that Marlin created confusion by offering 4 different interfaces with different functions!  There is Stock (CrealityUI), theirs (MarlinUI), Enhanced UI and JyersUI.  In JyersUI as in the CrealityUI, there is no this menu.  And for example in Enhanced UI or CrealityUI, there is no management of UBL with or without BLTouch as in JyersUI.  The JyersUI version is more efficient, for example in Manual Mesh (which is in fact the same as that with BLTouch but without BLTouch and therefore more precise in terms of auto bed leveling).

tititopher68-dev, 2021-12-23:
@benebrady 
Another clarification, since Marlin 2.0.6, it is no longer necessary to act on the Endstop software for the Z Offset.  This happens automatically and was introduced with Marlin for LCD 12864 screens under the name Wizzard Z-OFFSET.  This is obviously the case here by using the Z-Offset functions of the Prepare menu.