Jyers/Marlin #1722: Can't save/load UBL mesh with index other than 0

moriraaca, 2021-12-27:
### Description

Perhaps it's a known limitation, but I can't find any info: it seems that on my Ender3V2 running `E3V2-UBL-BLTouch-15x15-v4.2.2-v2.0.1.bin` I have only one UBL mesh slot available?

I can't choose other slots using LCD&Knob - when I enter "edit mode" and turn the knob nothing changes. And when I run `G29 S1` I get this:

```
Send: G29 S1
Recv: ?Invalid storage slot.
Recv: ?Use 0 to 0
Recv: ok
```

Is Ender3V2 limited to just one slot?

### Steps to Reproduce

Option (A)
1. Go to Level menu, scroll down to "Mesh Slot" and click it
2. Try turning the knob in both directions

**Expected behavior:** [What you expect to happen]
The number changes to something else than 0

**Actual behavior:** [What actually happens]
The number stays 0

Option (B)
1. Issue `G29 L0` and then `G29 S1` command

**Expected behavior:** [What you expect to happen]
Loaded mesh is saved in slot 1

**Actual behavior:** [What actually happens]
Error message is shown:
```
Recv: ?Invalid storage slot.
Recv: ?Use 0 to 0
```

tititopher68-dev, 2021-12-27:
@moriraaca  
It's normal behaviour with the v2.0.1. L'espace de l'EEprom est limitée et fixée arbitrairement à 48 bytes par Marlin. Plus il y a de fonctions, moins il y a d'espace disponible. Le nombre de slot est calculé par le code Marlin en fonction de l'espace disponible dans l'EEprom. Si vous prenez le binaire que j'ai fourni pour la v2.0.2g dans la zone discussion, vous allez retrouvé 2 slots parce que j'ai fixé à 64 bytes la taille de l'EEprom. On pourrait même en théorie aller jusqu'à 128 !

moriraaca, 2021-12-27:
Thank you, I guess this can be closed then.

closed 2021-12-27 by moriraaca