Jyers/Marlin #1730: Weird characters on scrolling file name

paul-aviles, 2021-12-29:
### Description
Just compiled the latest bugfix-JyersUI for an Ender 3 V2, only made changes to the bed size 220 x 220 and 5x5 BLTouch (i am using the CR Touch)

Had no issues really, except that the file name displayed at the top has some weird random characters appended at the end of the file as it scrolls to the left and at the right side margin. See photo attached. 

### Steps to Reproduce

Send a job to the printer using OctoPi and file name has weird random trailing characters. 

#### Additional Information

Attached are the  `Configuration.h` and `Configuration_adv

[Ender3V2-220x220-BLTouch5X5.zip](https://github.com/Jyers/Marlin/files/7787556/Ender3V2-220x220-BLTouch5X5.zip)
[image]

tititopher68-dev, 2021-12-29:
If you used the sources directly from Marlin, then yes!  no wonder you have this.  These sources already present a lot of issues and use the new HAL management which poses a lot of problems with a lot of equipment.  What appears is called a Screen Glitch due to a loss of serial communication between the motherboard and the screen.  I advise you instead to use the sources of v2.0.2g in the discussion area and with the configuration files of this v2.0.2g.

paul-aviles, 2021-12-29:
Thanks, let me give it a try.

paul-aviles, 2021-12-30:
@tititopher68-dev I recompiled using https://github.com/Jyers/Marlin/files/7769341/Marlin-bugfix-JyersUI-v202g.zip as the source, and the sample files under \Marlin-bugfix-JyersUI-v202g\Configuration Files\E3V2 Templates\BLTouch-3x3\ as config files so Configuration.h and Configuration_adv.h which only modified the  X_BED_SIZE = 220,  Y_BED_SIZE = 220 and GRID_MAX_POINTS_X  = 5.  

It compiled without any issues, but the Printer Kill Reason M112 Shutdown Restart Required message appeared again. For instance, I was on the printer and did a bed level, it generated Communication timeout while idle in serial. log
[serial.log](https://github.com/Jyers/Marlin/files/7791312/serial.log)

Not sure what, but then I used one of the pre compiled E3V2-BLTouch-5x5-v4.2.2-v2.0.2g.bin and seems to be working OK.  Hoping there is no really much difference so don't need to change myself anything specifically, but do wonder why the compile version I did would fail. 

Was I supposed to use the configs under \Marlin-bugfix-JyersUI-v202g\Configuration Files\E3V2 Templates\BLTouch-3x3,  the ones under v2.0.2.g\Configuration Files\Jyers's Config or the ones under Marlin-bugfix-JyersUI-v202g\Marlin-bugfix-JyersUI-v202g\Marlin ? Besides that, nothing else was changed.

Also, on a curious note, if I do a Z adjustment and try to make the nozzle go up or down it does not move at all. I need to select Live Move of whatever that option is called first, but the head moves to the left a few cm's and then I can do adjustments and micro steps up or down.

Best wishes for a prosperous and healthy 2022 folks!

Paul Aviles

paul-aviles, 2021-12-30:
So the serial errors were caused by a bad cable, specifically me trying to disable +5 v with tape so just ended up cutting the red cable and done. 

Just curious is the Z adjustment issue above is normal?

Paul Aviles

paul-aviles, 2021-12-31:
@tititopher68-dev even using the precompiled binaries from the zip file the garbage characters appear if the file name is long. There are no weird characters in the file name also so not sure, but this seems to be a bug.

tititopher68-dev, 2021-12-31:
@paul-aviles 
Isn't really a firmware issue, I mean this is not an interface level issue but still maybe an issue with the HAL and CPU managed by Marlin.  You must then open an issue on the GitHub of Marlin.  To understand, in the code of the UI, we only scroll the content of a variable.  It is the Marlin code which provides the content of this variable, code based on the serial communication between the SD reader and the CPU.

If you are on Octoprint, this can also be explained by a communication fault too.  Where there, it may be necessary to modify the speed in configuration.h from 115200 to 250,000 bauds.  There is also a parameter I can act on in configuration_adv.h, increasing the TX / RX Buffer. 

Jyers chose to base the firmware on the Bugfix Marlin branch.  But the observation that I have made is that since the introduction of the Marlin 2.0.9.0 base, we have lost a lot in stability and that does not get better with 2.0.9.3.  These are the users who do the tests and few issues are reported because many users use the customs firmware and Marlin does not take into account the issues in their own code if their code is used in these customs ... ubiquitous.