Jyers/Marlin #1731: * Estimated time for print does not seem to calculate properly *

mrrstrat, 2021-12-29:
### Description

With the 2.0.1 firmware a print job seems to not calculate the estimated time properly every time. I have been using RAFT print option  - during raft print the total print job looks to be a just few minutes long as reported on the monitor screen. After raft is printed the time calculation updates but often is not even close to the print job time. The print time hits zero and rolls over but remaining time appears to be correct after the roll over.

### Steps to Reproduce

1. New print job started from .gcode file
2. Observe time seems to be off
3. Rolls over at some point and job appears to end with time after roll over.

**Expected behavior:** [What you expect to happen]

Time calculation correctly calculates RAFT time plus print job time. Then decrements print activity from this total time (and possibly adjusts based on actual performance during print job.

**Actual behavior:** [What actually happens]

Time appears to be calculated but raft print time and print job do not appear to be correctly coupled - when print job time reaches zero and continues to print the time rolls over to a time that seems to reflect the reality of what is left to print.

#### Additional Information

1. I am compiling my own firmware - only making adjustments for mesh, CR touch, bed size, CR touch offset
2. Behavior has been observed on Ender3 V2 and Ender 3 Max with same code base (but slightly different compiled versions)
3. Job appears to finish at the final time calculation
4. Looks to happen only with BRIM and RAFT from what I can tell

mrrstrat, 2021-12-29:
I have not dug into the code nor did anything to debug issue. If this issue has not been experienced by anyone else then I am inclined to believe it is localized to my equipment and use.

If time calculation is either a known and unfixable issue or a known issue that is part of another issue in-process please close this issue or combine with current parent issue.

tititopher68-dev, 2021-12-29:
@mrrstrat 
Not a firmware issue, just look at #1150

mrrstrat, 2021-12-30:
Looks like a method issue - we can kill this issue.