Jyers/Marlin #1738: 2.0.1 UBL doesnt persist after a reboot. not saved to eeprom?

Behem07h, 2021-12-30:
E3V2-UBL-BLTouch-15x15-v4.2.2-v2.0.1.bin, the mesh created doesnt seem to save when pressing confirm on the "save to eeprom" prompt. 
Possibly (i dont really know much about custom marlin) due to "//#define UBL_SAVE_ACTIVE_ON_M500 //" being commented out in the firmware.

### Steps to Reproduce
1. Flash the firmware
2. Create a mesh, and save it to eeprom via the "save to eeprom" prompt when probing is complete. Wait for the beep indicating that it has finished saving to eeprom.
3. Turn off printer via the printer's on-off switch
4. Turn on printer, and check to see if the mesh is still there

**Expected behavior:** 
Mesh saves to eeprom and thus persists after rebooting the printer. 

**Actual behavior:** 
Mesh is not saved and does not persist after a reboot. Mesh is also inaccessible/unusable in print jobs because it does not exist in eeprom. Mesh is also unused in print jobs, even if they are done immediately after creating the mesh, without any reboots.

#### Additional Information

Video of the issue: https://drive.google.com/file/d/1_6hov58keoGgQrDEniKPKX946E8nT-Qe/view?usp=sharing

kbrownrigg, 2021-12-30:
I noticed the same thing last night. The mesh didn't stay between prints either

tititopher68-dev, 2021-12-30:
@kbrownrigg @Behem07h 
After every flash, you must to do a Reset to Default, following by a Restore Settings and following by a Store Settings to initialize correctly the EEprom . Golden rule.

I advise you, either to compile your own firmware with this source (and with the file's configurations from this sources !) or use a pre-compiled firmware in #1694 .
But be careful with your startGode : in UBL mode : 
G28;
G29 L0;
G29 A;
G29 J2;
And perhaps, in v2.0.1, the EEprom size is limited to 48 bytes and more features=more space in the EEprom used ! In the v2.0.2g, I extended the size of the EEProm to 64 bytes. Note that with this version a Reset to Default will be automatically made after flashing !

billygr, 2022-01-21:
Same issue here with E3V2-UBL-BLTouch-10x10-v4.2.2-v2.0.1.bin

Applied twice Reset to Default=> Restore Settings=>Store Settings

Using bedleveling plugin with:
G28       ; home all axes
M155 S30  ; reduce temperature reporting rate to reduce output pollution
M190 S65  ; (optional) wait for the bed to get up to temperature
G29 P1    ; automatically populate mesh with all reachable points
G29 P3    ; infer the rest of the mesh values
G29 P3    ; infer the rest of the mesh values again
@BEDLEVELVISUALIZER	; tell the plugin to watch for reported mesh
M420 S1 V ; enabled leveling and report the new mesh
M500      ; save the new mesh to EEPROM
M155 S3   ; reset temperature reporting
M140 S0   ; cooling down the bed

G29 T reports it correctly after running .

Should i give a try to the https://github.com/Jyers/Marlin/discussions/1694 ?