Jyers/Marlin #1764: BTT Smart Filament sensor false positives

Mickelin01, 2022-01-07:
### Description

BTT Smart Filament sensor worked fine under 1.3.5b. After upgrade to 2.0.1 I get frequent but random false positives, where the printer reports runout for no reason. I have tried with various values for the FILAMENT_RUNOUT_DISTANCE_MM: 0, 7, 10, 25.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.
[Mickes Config.zip](https://github.com/Jyers/Marlin/files/7829077/Mickes.Config.zip)

tititopher68-dev, 2022-01-07:
@Mickelin01 
Try with this evolution #1694 ...

ender3v2, 2022-02-02:
What do I have to enter on the display once I have installed the sensor?  ender 3v2

ramonskie, 2022-03-15:
i had the same problem.
i had to uncomment `#define FILAMENT_MOTION_SENSOR`

ruettenc, 2023-02-11:
i have the same issue using the BTT sensor

ruettenc, 2023-02-11:
> i had the same problem. i had to uncomment `#define FILAMENT_MOTION_SENSOR`

so you had to modify the gcode for that? has that solved the issue?

ramonskie, 2023-02-13:
its in `configuration.h`
and then you need to compile it.

but i moved over to klipper now works much more reliable now for me.