Jyers/Marlin #1769: Icons missing from LCD after update

Sonicboom247, 2022-01-10:
### Description

made a few tweaks to the firmware, recompiled and uploaded and now only see text but all icons are gone from TFT, tested all three DWIN_SET folders and none  bring back icons unless i go back to E3V2-ManualMesh-5x5-v4.2.2-v2.0.1.bin

### Steps to Reproduce

1. edit firmware
2. recompile
3. upload

**Expected behavior:**Icons visible on TFT

**Actual behavior:** All icons are missing

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

tititopher68-dev, 2022-01-10:
@Sonicboom247 
Because  your #define ICON in /src/lcd/e3v2/common/dwin_set is set to 7 instead of 9.

`#define ICON    7 // Icon set file 7.ICO` -> `#define ICON    9 // Icon set file 9.ICO `

Thank you Marlin for ignoring the usual DWIN_SET configuration for the E3V2 and modifying the config for their 'MARLINUI'!

Sonicboom247, 2022-01-10:
> @Sonicboom247 Because your #define ICON in /src/lcd/e3v2/common/dwin_set is set to 7 instead of 9.
>
> `#define ICON 7 // Icon set file 7.ICO` -> `#define ICON 9 // Icon set file 9.ICO `
>
> Thank you [...]

haha, thank you! i will adjust and test now, just upgraded the springs on the bed that seemed to be made of memory foam, so this is right on time.

Sonicboom247, 2022-01-10:
@tititopher68-dev thanks so much!, as you knew, that was it!.. much appreciated

Sonicboom247, 2022-01-10:
one last item, the bed leveling icon is still the Info Icon, however when i select Info i actually have bed leveling but no icons...lol is that because i selected Mesh bed leveling ?

tititopher68-dev, 2022-01-10:
@Sonicboom247 
Just for information, If you are using the Marlin bugfixes, you will still need to modify some Marlin source files (encoder.cpp, pause.cpp, settings.cpp, plateformio.ini). I am working in parallel on the continuation of v2.0.2h (# 1694) and mine v3.0.0a ... There are still things to modify in the sources since the very last bugfix marlin. I regret that Marlin did not go beyond the compilation tests ... a compilation without errors does not mean correct functioning ...

Also be careful if you do not start from the base of the configuration files provided here. The Manual Mesh cannot be configured as recommended by Marlin, which explains the problems in your interface.

And Reminder of the Golden Rule (unless you have correctly configured your configuration files and with the latest Bugfixes, the printer must indeed perform an automatic Reset To Default on first startup): after every flash, do a Reset to Default, following by a Store Settings and following by a Restore Settings.

Sonicboom247, 2022-01-10:
@tititopher68-dev i really like to stay up to date so sadly i am on the not bleeding edge of bugfix 2.0 but at least a few days behind, the latest fix they finally addressed was the blank screen for my Artillery Hornet and a few others, but then Level corners was broken....

as far as the automatic reset, i am taking advantage of EEPROM_INIT_NOW, however i don't think i remembered to enable it for the Ender, i also have two sidewinders, a Hornet and a Photon i try to maintain...oh and a Davinci Jr i am planning on gutting...lol

That being said, i  am going to go through and delete pretty much everything i have and start fresh with your configuration files if possible.

closed 2022-01-11 by Sonicboom247