Jyers/Marlin #1770: Auto bed Leveling BiLinear

Sonicboom247, 2022-01-11:
### Description
Auto Bed Leveling Bi Linear does not respect commenting out to use last measured hieght 
#define MANUAL_PROBE_START_Z 0.2  // (mm) Comment out to use the last-measured height
case MMESH_OFFSET:
            if (draw) {
              Draw_Menu_Item(row, ICON_SetZOffset, F("Z Position"));
              current_position.z = MANUAL_PROBE_START_Z;
              Draw_Float(current_position.z, row, false, 100);
            }
### Steps to Reproduce

1. set bed leveling to Auto Bed Leveling Bi Linear
2. comment out //#define MANUAL_PROBE_START_Z 0.2  // (mm) Comment out to use the last-measured height
3. compile for error
4. setting to Mesh Bed Level will accept and probing will go to the last measure height.

**Expected behavior:** no error and accept commenting out

**Actual behavior:** error and ABL will not accept use last measure height

#### Additional Information

Not sure if this is only for Mesh Bed leveling

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

tititopher68-dev, 2022-01-12:
@Sonicboom247 
Sorry but No issue ! In our file's configuration (folder Configuration Files / E3V3 Templates ) MANUAL_PROBE_START_Z is defined quite rightly in Configuration.h : 
```
#if ANY(MESH_BED_LEVELING, AUTO_BED_LEVELING_UBL, PROBE_MANUALLY)
  // Set a height for the start of manual adjustment
  #define MANUAL_PROBE_START_Z 0.2  // (mm) Comment out to use the last-measured height
#endif
```
Please, you are wrong. MANUAL_PROBE_START_Z is only use for ManualMesh Mode ( Noprobe [here ->probe manually] + AUTO_BED_LEVELING_BILINEAR ) in Jyers's firmware. In this mode, we have in the dedicaced menu 'Goto Mesh Value' (Case MMESH_OLD) to retrieve the last value if you do again a Mesh. And in addition, you have an additional menu 'Manual Tuning' to edit a Mesh (regardless of the mode) ....

Sonicboom247, 2022-01-12:
Perfect, thank you @tititopher68-dev , i had the feeling it was dedicated to ABL Bilinear, i will correct and update my config.

tititopher68-dev, 2022-01-12:
@Sonicboom247 
>#define MANUAL_PROBE_START_Z 0.2 // (mm) Comment out to use the last-measured height
and if you look at bedlevel.cpp : 
```
#if EITHER(MESH_BED_LEVELING, PROBE_MANUALLY)

  void _manual_goto_xy(const xy_pos_t &pos) {

    // Get the resting Z position for after the XY move
    #ifdef MANUAL_PROBE_START_Z
      constexpr float finalz = _MAX(0, MANUAL_PROBE_START_Z); // If a MANUAL_PROBE_START_Z value is set, always respect it
    #else
      #warning "It's recommended to set some MANUAL_PROBE_START_Z value for manual leveling."
    #endif
    #if Z_CLEARANCE_BETWEEN_MANUAL_PROBES > 0     // A probe/obstacle clearance exists so there is a raise:
      #ifndef MANUAL_PROBE_START_Z
        const float finalz = current_position.z;  // - Use the current Z for starting-Z if no MANUAL_PROBE_START_Z was provided
      #endif
```
This means that you would start from the current Z axis position (not that of an old value of a Mesh!), Which is not recommended, hence the warning `#warning "It's recommended to set some MANUAL_PROBE_START_Z value for manual leveling."`

At best if you really don't want to define MANUAL_PROBE_START_Z, then you just have to modify : 

```
case MMESH_OFFSET:
if (draw) {
Draw_Menu_Item(row, ICON_SetZOffset, F("Z Position"));
current_position.z = MANUAL_PROBE_START_Z;
Draw_Float(current_position.z, row, false, 100);
}
```

by : 

```
case MMESH_OFFSET:
if (draw) {
Draw_Menu_Item(row, ICON_SetZOffset, F("Z Position"));
#ifdef MANUAL_PROBE_START_Z
  current_position.z = MANUAL_PROBE_START_Z;
#endif
Draw_Float(current_position.z, row, false, 100);
}
```

Sonicboom247, 2022-01-12:
> @Sonicboom247
>
>
> ```
> #if EITHER(MESH_BED_LEVELING, PROBE_MANUALLY)
>
>   void _manual_goto_xy(const xy_pos_t &pos) {
>
>     // Get the resting Z position for after the XY move
>     #ifdef [...]

Thank you! that really explains it very clearly, i really enjoy using the last point in the mesh to adjust the next point. sometimes no change is needed and i can go through quickly instead of having to set a position to return to, but i like the Jyers UI so much i can deal with it either way.

closed 2022-01-12 by Sonicboom247

Sonicboom247, 2022-01-12:
compiled last night and flashed this morning, so far i have ALL menus including the Info has been replaced with Bed Leveling and all the bed leveling items are there! super happy to get this working (so far) , thank you!