Jyers/Marlin #1790: Bugs with Z-Offset menu

troglle, 2022-01-17:
I installed E3V2-Default-v4.2.2-v2.0.1.bin (firmware is stock, downloaded the precompiled file from https://github.com/jyers/marlin/releases/) with original icons pack (DWIN_SET (Original)). When try to tune the z-offset (from the z-offset menu) I encountered these bugs (Live Adjustment option is active - checked):
- First one - when changing the z offset option (in the menu) from positive to negative values and then back to 0, the 0 value change to - 0.00 (minus, before it was 0.00)
- Second one - when tune the z offset with Microstep Up/Down, after several clicks these options stop working - z offset values are changing, but the stepper motor isn't turning. I need to enter the menu again to fix this.
- Third one - when try to home z axis (from the option in the menu) the axis is homing and then the hot end and bed go to opposite side from end switches. I had to turn off the printer to prevent damage.

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

tititopher68-dev, 2022-01-17:
@troglle 
Just use #1694 version

MikeOxawopper, 2022-02-08:
i Have similar issues with my install, im sure it was better a couple of months ago but i had an issue with the nozzle becoming too close so i tried it from the manual menu but although the numbers were scrolling no noticeable difference was seen in height.

Secondly the manual values were wildly different to the auto mesh by up to 1/2mm, having re-auto meshed twice the z height checked but was too high like a dog laying a cable! paper test was perfect but to get it to print i had to baby step from -2.80  to -3.18.

I used to be able to manually dial x y z when live was enabled but that appears to do nothing either, only moves with dial and click.
I can print the same file 3 times one will be too high, one too low but if i tune low z to the high z or the high z to the low z value it will work?

tititopher68-dev, 2022-02-08:
@MikeOxawopper 
Please be more specific in your information. Firmware version? etc...
I can assure you that this is not a firmware issue. my Z-Offset remains perfectly stable. On the other hand many physical parameters can intervene... not to mention the methodology. A Z-Offset must be done (like everything related to Bed Leveling) at printing temperature with a clean Bed and a clean nozzle. The surface of the Bed evolves over time over the impressions, the springs also with the temperature, the nozzle also, the Vslots also etc... in general after 100h of printing, it is advisable to make a revision of all the physical parameters and adjust them...

tititopher68-dev, 2022-03-09:
@ddudek 
Again, prefer version v2.0.2i in #1694.  Then don't forget to always make your settings at printing temperature for the Nozzle and the Bed.  Including for the Mesh and wait at least 5 good minutes once the Bed has reached temperature so that it is homogeneous over the entire surface.  Of course, there remains the human factor for the use of the sheet of paper.  Taking all this into account, a difference of 0.4 mm is quite possible, not to mention (and from experience) other faults in the geometric settings of the printer...in any case, the firmware cannot be blamed,  only the basic Marlin code, common to all printers, manages the application of the Z-Offset...
Finally, do not make a comparison with Creality firmwares that use older versions of Marlin!

For my part, I have no variations of Z-Offset either in Bilinear or UBL with a perfect 1st layer...