Jyers/Marlin #1791: Two different progress % values sent to and diplayed on printer LCD

nielsrahn, 2022-01-17:
Hey guys

Running **E3V2-UBL-BLTouch-15x15-HS-v4.2.2-v2.0.1** and **OctoPrint 1.7.2, Python 3.7.3, OctoPi 0.18.0**.

I am having some strange issues with the progress % on the printer LCD. I have the following three plug-ins installed as recommended per Jyers wiki:

M73 Progress
Detailed Progress
DisplayLayerProgress

The progress % seems to jump back and forth from the correct % as also stated in OctoPrint (with PrintTImeGenius installed too) and a wrong value, often some % behind.

I've analysed the sent messages, specially M117 and M73 - there are alternatively different % values sent to the printer via M117, but I am not sure, which ones are correct or wrong.

Anybody have an idea how to solve this?

Many thanks!

tititopher68-dev, 2022-01-17:
@nielsrahn 
Not a firmware issue.
Any ! They are extrapolated in different ways... It is only Octoprint that calculates the progress in your case, the firmware only displays the value received by Octoprint without modifying it.

nielsrahn, 2022-01-18:
> @nielsrahn Not a firmware issue. Any ! They are extrapolated in different ways... It is only Octoprint that calculates the progress in your case, the firmware only displays the value received by Octoprint without modifying it.

Thanks for your reply! That of course makes sense.

So in my case, there are probably two different progress values sent to the printer - maybe from two parallel running Plug-Ins?